jc-kynesim/rpi-ffmpeg #11: Advice for fewer calls in frame refresh?

Fredrum, 2021-09-04:
Hi! Iv'e now got a test program based on copying select parts of hello_drmprime and egl_vout and I'm starting to look at how I can make some optimisations to hopefully lower cpu cost.
I know your egl_vout device was just for testing and not made for performance!

In my other v4l2 test program that does a drm pipeline for the Picamera I have only this in my main program loop,

```
ioctl(fd, VIDIOC_DQBUF, &buf);
glEGLImageTargetTexture2DOES(GL_TEXTURE_EXTERNAL_OES, dma_image);
ioctl(fd, VIDIOC_QBUF, &buf);
```
 and I'm able to avoid doing anything more for GL texture refresh for each frame. Also the 'fd' is not changed per frame.
 
 So I was wondering if it would be possible to do something similar with your avcodec pipeline?
 Maybe these are too different api's, stateful vs stateless? And they just don't operate the same way?
 
 Anyhow, I'm thrilled to have code that decodes hevc on the hardware and I'm sure to add it to my other project!
 
 Cheers!

jc-kynesim, 2021-09-06:
Yes you can just refresh the contents and that works - when I last looked I was surprised by how little CPU it saved. Beware that in a set of async operations like the above where the the texture render is deferred (I believe that call will return before the render is complete) and you return the buffer to decode before you know it has completed you can get odd results where the decoder overwrites the buffer before GL has finished rendering it.  This also applies to DRM output (which is why there is a little Q).
In terms of CPU usage remember that the default options run both the decode & the display as fast as they will go which inevitably gives high CPU.  If paced the numbers are better.  "-show_all 1" before the "-f vout_egl/drm" should give you 1 frame / v-sync.

Fredrum, 2021-09-06:
Ok that's useful to know maybe I won't spend time trying to do a 'just refresh contents' setup yet If its not saving much. My test program has the fake 'usleep(3000)' timing (as my target use should already have the pacing set up) and is running at ~8% for the process for a total of 12% for the system, as read in the task manager.

Can I ask about what is the function of the [32] sized egl_aux_t ?
For an experiment I changed:
```
	for (i = 0; i != 32; ++i) {
	    if (fd_tex[i].fd == -1 || fd_tex[i].fd == desc->objects[0].fd) {
		da = fd_tex + i;
		break;
	    }
	}
```
to 
`for (i = 0; i != 1; ++i) {`
and it still seemed to play the video fine but also still ran a loop cycling over multiple fd's.
Are they part of the queue you mentioned? Or are you decoding smaller parts of a full frame?

jc-kynesim, 2021-09-06:
Hi

> Ok that's useful to know maybe I won't spend time trying to do a 'just refresh contents' setup yet If its not saving much. My test program has the fake 'usleep(3000)' timing (as my target use should [...]

Thats redundant code that I should remove - its from when I used to
track dmabuf fds so I didn't have to remap them every time - just like
you were suggesting!

Fredrum, 2021-09-06:
Oh ok cool so I can remove that, but not that it took any time to run.

Do you know if this new ffmeg is going to be included in a Bullseye release for the Raspberry?
Raspberry OS or Ubuntu? Or will it be with a later update do you think?

jc-kynesim, 2021-09-06:
This will be in Pi OS Bullseye. (4.3.2 is the release that Bullseye uses)

closed 2021-09-10 by Fredrum