jc-kynesim/rpi-ffmpeg #13: Advice for latency? And difference in cpu use h264/hevc?

Fredrum, 2021-09-15:
Now Iv'e managed to use this new ffmpeg/avcodec version in the net streaming application I working in,  https://github.com/Fredrum/chiaki   but so far I am not able to match the low latency I get using omx/ilclient so I was wondering if you had any advice for getting the lowest possible latency?

Another thing is I'm noticing that the h264 decode runs a bit more sluggish and using more CPU than the hevc decode. 
For the same 40mbit jellyfish file I get around 8% cpu with hevc and 22% cpu with h264. (Using this: https://github.com/Fredrum/hwdectogl/blob/master/glHevc.c)
In comparison I get around 2-3% using omx/ilclient.

Are those roughly the numbers that I should expect do you think? Should it be that big difference? Just wondering so I know when to stop trying to find errors in my own code. :)

I wanted to get this working so I could get the program (Chiaki) to run on 64bit and on non Raspberry OS's  :)

Cheers!

jc-kynesim, 2021-09-15:
If your OMX client uses opaque buffers and HVS display directly then I'm not at all surprised by the CPU difference - that shunts all the processing to the VPU so all the ARM is doing is marshalling stuff. V4L2->GL is much more ARM intensive. I think you've probably got it close to as good as it gets. I'm not sure how you can improve latency.

Fredrum, 2021-09-15:
Ok cool thank you that's great to know!
Is v4l2 still being worked on or is that 'frozen' code?
Anyway that's great next up for me is to try all this probably on Ubuntu 64bit! 

Cheers!

jc-kynesim, 2021-09-15:
I wouldn't go as far as saying that the V4L2 code is frozen, but I wouldn't expect anything much more by way of performance improvements.

Fredrum, 2021-10-01:
It looks like I'm getting RGB888 when using AV_PIX_FMT_DRM_PRIME. Does that mean that there's some YUV -> RGB conversion happening somewhere? And is that done in software?
If it is could that be bypassed so that I could do that calculation using the GPU instead? Which should be faster I think.

closed 2021-10-01 by Fredrum