jc-kynesim/rpi-ffmpeg #15: Is drm render not possible with h264?

Fredrum, 2021-09-29:
Hello!
I was planning on using this decoding both h264 and hevc and piping to drm render under CLI.
(and also have an option for X11+GL)
Now when I'm trying I'm getting error message saying drm is not available with h264. I don't remember this error from before but maybe I never tried that combination.

Is this really the case or just a limitation of 'hello_drmprime'? Seems a bit unlikely that you could do one and not the other?

Cheers!

jc-kynesim, 2021-09-29:
hello drmprime just does H265. It could be quite easily extended to do H264 but its lacking the codec selection logic to allow it to notice it needs h264_v4l2m2m rather than the default s/w decode.  Its display stage needs the frames already in a dmabuf which doesn't occur otherwise.

Fredrum, 2021-09-29:
Ok great!  So that would be in 'drmprime_out.c' or in 'hello_drmprime.c'?
It sounds like I might be able to have a go with that.

Fredrum, 2021-09-30:
Yesh!  I got it to work!  It was easier than I thought.
I just followed the errormessage and then realised what you meant. And I had my own test code for openGL based on your examples and I just did what I did in that one.
Great!

closed 2021-09-30 by Fredrum