jc-kynesim/rpi-ffmpeg #16: Is there a magic GPU and CMA memory setting?

Fredrum, 2021-09-29:
Hi!

I'm getting a lot of memory errors trying to decode a 40mbit 1080p60 video stream and have been trying to find a combination that works. (On my 4GB RPi4b) Seems tricky!  :)

Is there a magic combo that works?

Cheers!

jc-kynesim, 2021-09-29:
This is H265 / H264 / other?  In ffmpeg or another prog?  If you are stopping and restarting the H265 decoder a lot without closing the process then I fixed a bad memory leak about a week ago.

Fredrum, 2021-09-29:
This was with both codecs I think. But h264 seemed more memory hungry than hevc, which I guess makes sense.
And sorry yes using both hello_drmprime and my small (not well coded) ffmpeg/avcodec test program.
Which branch did you do the fix in?

closed 2021-12-15 by jc-kynesim