jc-kynesim/rpi-ffmpeg #19: something causing errors building kodi from master with 4.4/rpi_import_1

graysky2, 2021-11-06:
I can build the latest kodi master commit without issues, but when I apply your code from https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.4/rpi_import_1, I get a build error.

Build log with error: https://gist.github.com/graysky2/505f522102ab8c106337a315c64bd287

I made the patch to apply like this:
```
git checkout dev/4.4/rpi_import_1
git pull --rebase
git fetch xbmc
git diff xbmc/release/4.4-kodi > /tmp/nexus.patch
```

jc-kynesim, 2021-11-08:
Try diffing from n4.4 - my dev patches are normally based from the branch point for ease of automated patch gen - I may then merge with the release branch to produce a release/version/name branch.  Your log seems to have been truncated before whatever error it is you had.

graysky2, 2021-11-08:
>Try diffing from n4.4

I will try it and report back.

> Your log seems to have been truncated before whatever error it is you had.

It is not actually... the github interface tries to save bandwidth by showing the first somany lines.  Click the [view full file here](https://gist.github.com/graysky2/505f522102ab8c106337a315c64bd287/raw/6195541dd78bd75fdadbd97b355684d971a61140/no%2520distcc%2520build.log) link at the top to see it in its entirety.

graysky2, 2021-11-08:
@jc-kynesim - Generating the patch from n4.4 did not help.  Seems to have errored out at the same place (building RenderManager.cpp).

https://gist.github.com/graysky2/151fa36f71e0ade0b823687df0f3cb49

jc-kynesim, 2021-11-08:
Does it build without the ffmpeg patch?  I've no idea what the root cause of that error deluge is - do you?

graysky2, 2021-11-08:
Yes, if I remove the ffmpeg patch, the build ends successfully.  I have no insights into this one.

graysky2, 2022-08-21:
I should update this, the issue is no longer present.
```
git checkout dev/4.4/rpi_import_1
git pull --rebase
git fetch xbmc
# do not build this way: git diff xbmc/release/4.4-kodi > /tmp/nexus.patch
git diff n4.4 >/tmp/nexus.patch
```

closed 2022-08-21 by graysky2