jc-kynesim/rpi-ffmpeg #20: Branch test/4.3.2/rpi_main issues with hevc playback on a 4k display

sava41, 2021-11-08:
Hi, 

Were trying to use the test/4.3.2 branch to play 4k hevc and 2k h264 files on a 4k display. We've successfully ported ffplay to use drm_prime and opengl (with both hevc 4k and h264 2k) on a 1080p display. Unfortunately, the hevc decoder seems to fail when playing the media on a 4k display:

`[hevc @ 0xb3d1d890] Hwaccel V4L2 HEVC stateless V1; devices: /dev/media0,/dev/video19
Failed to alloc 12443648 from dma-heap(fd=10): 12 (Cannot allocate memory)
[hevc @ 0xb3d749c0] v4l2_request_hevc_start_frame: Failed to get dst buffer
Failed to alloc 12443648 from dma-heap(fd=10): 12 (Cannot allocate memory)
[hevc @ 0xb3d749c0] v4l2_request_hevc_end_frame: Failed to get dst buffer
[hevc @ 0xb3d749c0] hardware accelerator failed to decode picture
[hevc @ 0xb3d1d890] Could not find ref with POC 14
`

I have also tried the hello_drmprime example using the test/4.3.2 compiled libraries and it works with the 4k display. As this doesn't seem to be an issue with any of the opengl api calls, what could be going wrong?
 
Any pointers would be greatly appreciated.

jc-kynesim, 2021-11-09:
Before addressing your point I need to point out that you are doomed. The Pi h/w doesn't have the power to render to a 4k window in real time via gl.  An HD window is approx as good as it gets. (The limitation is more on display size rather than video size, but both matter depending on what you are doing)
Otherwise 4k frames use a lot of memory & ffmpeg tends to assume that it is unlimited, but it really isn't. For this purpose It comes out of the CMA pool - you can alloc 512M from config.txt but more than that is unreliable I believe. GL also uses substantial quantities of that pool for its store so life becomes even worse when using that.
Sorry.

sava41, 2021-11-09:
I'm a little confused then because we have no issues creating a 4k window. For clarification were running without x server and creating the window through gbm. 2k video upscale on a 4k window plays with no issues.  KMScube also runs with no issues at a fast framerate at 4k.

Also when you say setting cma to 512M in config.txt you mean the setting `cma=512M`?

jc-kynesim, 2021-11-09:
OK - I've only tried with X in the way and that definitely doesn't work.  To set cma in config.txt you need cma-512 on the end of the "dtoverlay=vc4-kms-v3d" line, thus (cma = xx on its own line does nothing):

dtoverlay=vc4-kms-v3d,cma-512

If it works for you through gbm then hurrah! and maybe just a bit more CMA will do the trick.

jc-kynesim, 2021-11-09:
A couple more thoughts:
1) The H264 decoder produces YUV or NV12 - this can be fairly directly consumed by GL, the H265 decode produces SAND8 (for 8-bit) or SAND30 (for 10-bit).  We have code for SAND8 texture import in GL assuming a current enough mesa, SAND30 is more tricky and when last seen we didn't have that.
2) If you do make this work I'd be very interested in seeing your code!