jc-kynesim/rpi-ffmpeg #21: Copying frames from HEVC block to H264 block impossible due to incompatible pixel format?

mikk150, 2021-11-15:
Just a question...

So as I understand, HEVC v4l2 device outputs frames in SAND format, and H264 expects in YUV format, therefore we could not use RPI HW blocks to transcode HEVC to H264?

If that is so, that is a bummer, as I literally bought RPI for that reason...

jc-kynesim, 2021-11-15:
That is certainly the state of the art. It may be possible to get the V4L2 H264 encode (which is what I assume you are hoping for) to accept SAND8 (which is what H265 8-bit video will decode to), but there is no chance of getting it to accept SAND30 (output from10-bit H264 video). You can convert between the two in s/w on the ARM but that is (relatively) slow.

nyanmisaka, 2021-11-16:
AFAIK the `/dev/video12` wrapper for `OMX.broadcom.isp` is the image processor inside the Pi 4 SoC, does it support pixel format conversion? If true we can add it as a filter to convert SAND30 to SAND8 before sending to the V4L2 encoder.

jc-kynesim, 2021-11-16:
Sadly the only h/w that can consume (and convert) SAND30 on a Pi4 is the HVS. Under KMS that is given to the exclusive use of the display. In _theory_ you can get get DRM (owner of the HVS) to render to a special connector rather than the screen and that would do what you want, but I've yet to see that done and getting hold of a DRM lease/fd that will let you do even try requires (AFAIK) whoever is holding it at that time (the display) to give it up which probably isn't what you want as it will take the display away.

jc-kynesim, 2021-11-16:
(under MMAL this was all relatively simple, in the brave new advanced standardized linux world it is all but impossible)

closed 2021-12-15 by jc-kynesim

mikk150, 2022-08-22:
> Sadly the only h/w that can consume (and convert) SAND30 on a Pi4 is the HVS. Under KMS that is given to the exclusive use of the display. In _theory_ you can get get DRM (owner of the HVS) to [...]

BTW, if it took blanking display to transcode HEVC to H264, it would actually work, as many people who want it I presume are trying to use RPI as transcode server(as it is cheap, power efficient and can decode HEVC 10bit and encode H264)

jc-kynesim, 2022-08-25:
Since I last commented on this I've done more work on writeback connectors and I can see how this might work, but don't hold your breath as its not trivial and, with my current understanding, could only be started from outside X as I haven't worked out how to get a DRM fd that will let be get at writeback from within X.  But assuming that I built a filter that would let you use that that, annoyingly, only gets you SAND30->RGB as RGB is all that the HVS will output, so then you need the ISP to convert that back into YUV and that extra pair of conversions I suspect would use enough memory bandwidth to break real-time 1080p60 transcode.

BTW for what it is worth, h/w 8-bit H265 -> H264 transcode does now work.

mikk150, 2022-08-25:
> could only be started from outside X as I haven't worked out how to get a DRM fd that will let be get at writeback from within X

would be fine, because transcode servers do not even need X(Or any working output for that matter, could possibly even disable HDMI output, and people would still rejoice)

> I suspect would use enough memory bandwidth to break real-time 1080p60 transcode.

Bummer, as UHD 10bit to FHD is really far out of reach then

> BTW for what it is worth, h/w 8-bit H265 -> H264 transcode does now work.

does it do UHD H265 to FHD H264? if so, this already is really big stepping stone

jc-kynesim, 2022-08-25:
Sorry to get you excited - the working transcode is just 8-bit unscaled. I've not built the v4l2 ISP rescale filter yet, but I'll put that on my todo.

mikk150, 2022-08-25:
> Sorry to get you excited - the working transcode is just 8-bit unscaled. I've not built the v4l2 ISP rescale filter yet, but I'll put that on my todo.

Nah, I'm not exited, I know that there are hardware limitations in place (and why this is really hard, possibly impossible) to do. So I am not holding my breath.. Just a thing that would be really awesome to see(Get over hardware limitations in possibly hacky ways)

nyanmisaka, 2022-08-25:
Above 30fps is still acceptable for on demand video transcoding (4k hevc 10bit -> 1080p h264 8bit), considering that most users' video source is 23.98fps.