jc-kynesim/rpi-ffmpeg #22: [dumb question] configure flags to enable hevc (and h264) v4l2 m2m decode?

egnor, 2021-11-16:
Hi! This is a "how do I" rather than a real bug report, so feel free to delete it if your spoons are limited (and sorry for the bother).

I have an RPi 4 with `dtoverlay=vc4-kms-v3d` and `dtoverlay=rpivid-v4l2`, recently updated firmware, and an updated Raspberry Pi OS. When I compile `hello_drmprime` against the system libraries, it runs, though it prints this:

```
% ./hello_drmprime jellyfish-3-mbps-hd-hevc-10bit.mkv
No connector ID specified.  Choosing default from list:
Connector 32 (crtc 76): type 11, 1920x1080 (chosen)
Connector 40 (crtc 0): type 11, 0x0
[hevc @ 0x764650] Trying V4L2 devices: /dev/media2,/dev/video19
drmWaitVBlank failed: Invalid argument
drmWaitVBlank failed: Invalid argument
... [repeating once for each frame(?)] ...
```

That's fine and dandy. _However_, I'm trying to build directly from your tree using a conan (conan.io) recipe. (Don't worry about the conan part, just assume I'm building by hand.) I'm pulling from this github repo's `release/4.3/rpi_main` branch, and making sure to add `--enable-libv4l2` `--enable-rpi` `enable-sand` `--enable_v4l2_m2m` (all the relevant configure flags I could find).

But when I build hello_drmprime against the library built that way, I get this:

```
% build/hello_drmprime jellyfish-3-mbps-hd-hevc-10bit.mkv 
No connector ID specified.  Choosing default from list:
Connector 32 (crtc 76): type 11, 1920x1080 (chosen)
Connector 40 (crtc 0): type 11, 0x0
Decoder hevc does not support device type drm.
```

And then it exits and obviously nothing gets shown. So, I think it's building but somehow not including support for the hevc_v4l2m2m decoder? The [config.log](https://github.com/jc-kynesim/rpi-ffmpeg/files/7549115/config.log) is attached if that helps any.

Thanks for any help you can give (and for all your work on this)!! :pray:

jc-kynesim, 2021-11-17:
If you look at pi-util/conf_native.sh you'll find the config flags I use. I just use that script to setup my builds.

closed 2021-12-15 by jc-kynesim

egnor, 2021-12-16:
> If you look at pi-util/conf_native.sh you'll find the config flags I use. I just use that script to setup my builds.

Sorry, where is `pi-util/conf_native.sh`?

egnor, 2021-12-16:
(Breadcrumb for anyone else who struggled with github search, which only looks at `master` by default: https://github.com/jc-kynesim/rpi-ffmpeg/blob/release/4.3/rpi_main/pi-util/conf_native.sh)

egnor, 2021-12-16:
@jc-kynesim - is this believed to work without --mmal?

I can configure and build the libraries, but attempting to link a program against them generates a ton of link errors saying that all the various mmal-related system library functions aren't available. Could be some kind of error on my side, if you believe it works fine without --mmal then I'll keep debugging.

jc-kynesim, 2021-12-16:
I've just checked with hello_drmprime and yes it does work for me. FYI this was clean builds of both ffmpeg & hello_drmprime - you probably need to make sure you have no old objects or libraries lying around. I have a little script for wiping old libs "sudo pi-util/clean_usr_libs.sh".
On the other hand I haven't tried actually removing mmal libs from anywhere to check they are really no longer needed (they certainly aren't on the hello_drmprime command line)

egnor, 2021-12-16:
Thanks!

For the record (and anyone else walking this path), I found my problem -- I'd been passing `--enable-rpi` even when not wanting to use MMAL and other Broadcom-proprietary libraries. Turns out that triggers a dependency on MMAL. To just use V4L2 and other open APIs, it seems `--enable-rpi` is not necessary or wanted. (Correct me I'm wrong!)

jc-kynesim, 2021-12-17:
That is correct --enable-rpi was my original catch-all switch for Pi3 work and enables stuff with rpi in the name (in general).  All of that has broadcom related requirements even if it is only for memory allocation (a lot of that work is pre easy dmabuf, and the legacy cache ops are still much better than anything you can do directly under linux).