jc-kynesim/rpi-ffmpeg #26: Spurious (?) "Unexpected input whilst draining" warning with h264_v4l2m2m

egnor, 2021-12-21:
While it's relatively harmless, I see an "unexpected input whilst draining" while completing an h.264 video using h264_v4l2m2m on the Pi:

`[h264_v4l2m2m @ 0x2e50a00] Unexpected input whilst draining`

As far as I know I'm not providing anything extraneous after the empty packet that signals EOF. This seems to come from [this part of v4l2_m2m_dec.c](https://github.com/jc-kynesim/rpi-ffmpeg/blob/release/4.3/rpi_main/libavcodec/v4l2_m2m_dec.c#L388):

```
    if (s->draining) {
        if (!s->buf_pkt.size) {
            av_log(avctx, AV_LOG_WARNING, "Unexpected input whilst draining\n");
            av_packet_unref(&s->buf_pkt);
        }
        return NQ_DRAINING;
    }
```

I'm not an expert at ffmpeg internals, BUT... shouldn't the `if (!s->buf_pkt.size)` be `if (s->buf_pkt.size)` instead? (That is, a NON-empty packet is surprising while draining.)

jc-kynesim, 2021-12-22:
That logic seems sound to me. Wonder why I've never seen that? Either way I think you are right.

jc-kynesim, 2021-12-24:
Fixed - thanks

closed 2021-12-24 by jc-kynesim