jc-kynesim/rpi-ffmpeg #27: [question] is it possible to *recycle* DRM "PRIME" (aka dma-buf) buffers into an avcodec decoder?

egnor, 2021-12-24:
Hi,

With the right magic (as demonstrated nicely in hello_drmprime.c!) I can get libavcodec to manage the hardware codec interfaces and deliver me frames as dma-buf file descriptors with some metadata, which I can then use to set up DRM framebuffers, yay cool.

What I'm wondering is whether I can deliver those buffers back to the codec for reuse, to avoid needing to re-do the buffer allocation, conversion to GEM buffer objects, assignment of DRM framebuffers, and all that jazz?

Or... maybe doing all that is superfast in comparison to frame decode and display times so I just shouldn't worry about it?

jc-kynesim, 2022-01-04:
dmabuf allocation is pooled in the decoder so when you unref the ffmpeg avbuf then that returns to the pool - no reallocation required.  It might well be nice to attach BO handles to the same structure to avoid reconversion but that is fast & I've been trying to avoid reworking that bit of the pooling code.
So - in short: don't worry

egnor, 2022-01-04:
Sweet! I shall not worry and/or be happy!

closed 2022-01-04 by egnor