jc-kynesim/rpi-ffmpeg #29: AVFrame from h264_v4l2_m2m has "key_frame" set incorrectly

egnor, 2022-01-04:
Running `jellyfish-3-mbps-hd-h264.mkv` through h264_v4l2_m2m, almost all of the resulting `AVFrame` structs have `key_frame` set to 1 for some reason, even though an independent dump of the file shows only an occasional key frame as one would expect. (Looking at the code, it seems legit enough, so I'm not sure why it's working.)

If this is surprising, I can work on putting together a repro case by adding some logging to hello_drmprime.c.

Also, that codec seems to not set `pict_type` at all, even though deriving it from the KEYFRAME / PFRAME / BFRAME buffer flags should be fairly straightforward.

egnor, 2022-01-04:
Actually I think I have a theory about why `key_frame` is set wrong. In [ff_v4l2_buffer_avpkt_to_buf_ext()](https://github.com/jc-kynesim/rpi-ffmpeg/blob/release/4.3/rpi_main/libavcodec/v4l2_buffers.c#L785) this happens:

```
    if (pkt->flags & AV_PKT_FLAG_KEY)
        out->flags = V4L2_BUF_FLAG_KEYFRAME;
```

However, the packet does _not_ have `AV_PKT_FLAG_KEY` set, then `out->flags` is untouched. As far as I can tell, buffers are recycled without being zeroed (!) so as buffers cycle through the kernel, this flag only gets set, never un-set, which means eventually most of the buffers have the flag set. And if it's set incorrectly on input to the kernel, it comes out incorrectly on output from the kernel. Probably the buffers ought to be zeroed somewhere before being reused? Or at least have the `flags` value rewritten.

Tangentially, [ff_v4l2_buffer_avframe_to_buf()](https://github.com/jc-kynesim/rpi-ffmpeg/blob/release/4.3/rpi_main/libavcodec/v4l2_buffers.c#L684) seems to have the keyframe logic backwards:

```
    out->buf.flags = frame->key_frame ? (out->buf.flags & ~V4L2_BUF_FLAG_KEYFRAME) : (out->buf.flags | V4L2_BUF_FLAG_KEYFRAME);
```

jc-kynesim, 2022-01-04:
I'll look into this.  Its worth fixing the keyframe flags, but they shouldn't be needed for the v4l2 decode to get the result right.  And as you say avframe_to_buf is just wrong so I'll definitely fix that.

jc-kynesim, 2022-01-04:
By the way - why do you care?  I'm not suggesting for a moment that this shouldn't be fixed but I just can't think of a use case for the info.

egnor, 2022-01-04:
Haha it's definitely minor and I was sort of wondering if you'd ask lol.

BUT for my application (video playback for immersive entertainment, e.g. escape rooms) being able to do precise seeks is important. It's up to the user to add keyframes at the right points in their videos but the system will report where they are so the user can verify that their seek points are where they expect them to be.

As far as pict_type I have no special reason for that except I enjoy seeing them go by lol. I was mostly looking at that when trying to debug the keyframe boolean.

jc-kynesim, 2022-01-04:
OK - fix pushed to dev/4.4/rpi_import_1 - I'll update the 4.3.2 branches in a bit. The decoder seems to just be copying flags from src to dst rather than actually setting them based on frame type so PTS would have done just as well. I now set pict_type from the V4L2 flags but given that the P & B flags are never set by the Pi decode, the only type you'll see is I.

closed 2022-08-02 by jc-kynesim