jc-kynesim/rpi-ffmpeg #30: h264_v4l2m2m and hevc codecs set AVCodecContext::pix_fmt differently

egnor, 2022-01-05:
Very likely I just don't understand the ins and outs of these values?

When using h264_v4l2m2m,
AVCodecContext::pix_fmt = AV_PIX_FMT_DRM_PRIME
AVCodecContext::sw_pix_fmt = AV_PIX_FMT_YUV_420P
AVFrame::format = AV_PIX_FMT_DRM_PRIME

When using hevc,
AVCodecContext::pix_fmt = AV_PIX_FMT_YUV_420P
AVCodecContext::sw_pix_fmt = AV_PIX_FMT_NONE (I think)
AVFrame::format = AV_PIX_FMT_DRM_PRIME

How are these values supposed to be set? Looking at the AVCodecContext (or some other non-frame-specific structure) how would I know whether to expect AV_PIX_FMT_DRM_PRIME, and if so what its "contained" pixel format would be? Or do I just have to wait for the first frame?

jc-kynesim, 2022-01-06:
Fair enough - the hevc one should be set more like the h264 one (not really sure why it isn't - but I'll check), but your best bet is definitely to wait for the 1st frame and unpack the frame descriptror.

jc-kynesim, 2022-01-06:
In my testing the avctx stuff is set correctly after the 1st frame (hevc). The hw decode init that sets the formats isn't called till after the getformat callback (and can't be) so you aren't going to get the pix_fmt / sw_format set "correctly" in there, but it will be set correctly when control returns to the main loop after that.

egnor, 2022-01-06:
Ah I didn't realize get_format() happened on first frame, I thought it happened on init. That explains some other stuff actually. (And h264_v4l2m2m is different because that codec is a specialized one that only supports the hardware. Probably in the fullness of time that would be better as something negotiated within the main h264 codec, as with hevc?)

Thanks for the explanations!

closed 2022-01-06 by egnor