jc-kynesim/rpi-ffmpeg #31: Failing to build with vout_drm and vout_egl (Raspberry Pi OS lite - bullseye 32bit)

careyer, 2022-01-23:
Hey @jc-kynesim ,

thank you very much for the awesome work you've done here.
For days and days I am trying now to build a version of rpi-ffmpeg which features the **vout_drm** and **vout_egl** output devices. (Yes I know they are experimental but... see end of msg ;-)
But whatever I end up with does not include these modules when running `ffmpeg -devices`

My system is:
- Raspberry Pi 4B (4GB)
- latest Raspberry Pi OS lite - bullseye 32bit (Release date: October 30th 2021, Kernel version: 5.10)
- rpi-ffmpeg: dev/4.4/rpi_import_1

/boot/config.txt contains:
```
#dtoverlay=vc4-kms-v3d
dtoverlay=rpivid-v4l2
dtoverlay=cma,cma-size=402653184

```
My build process is as follows:
```
# 0. Update system
sudo apt update
sudo apt upgrade

```
```
# 1. Install kernel headers
sudo apt-get install raspberrypi-kernel-headers

```
```
# 2. Install build dependencies for ffmpeg
# open /etc/apt/sources.list and uncomment all the deb-src entries
sudo apt update
sudo apt build-dep ffmpeg

```
```
# 3. create build directories
BUILD_DIR=~/build
mkdir -p $BUILD_DIR
cd $BUILD_DIR

```
```
# 4. Build modified ffmpeg by jc-kynesim
git clone https://github.com/jc-kynesim/rpi-ffmpeg.git
cd rpi-ffmpeg
git checkout dev/4.4/rpi_import_1
./configure --extra-libs=-latomic --extra-cflags=-I/usr/include/libdrm --disable-mmal --disable-thumb --enable-static \
--disable-stripping --enable-sand --enable-v4l2-request --enable-hwaccel=hevc_v4l2request --enable-decoder=hevc \
--enable-decoder=h264_v4l2m2m --enable-decoder=hevc_v4l2m2m --enable-libdrm --enable-libudev --enable-shared \
--enable-libx264 --enable-gpl --enable-opengl --enable-avcodec --enable-decoder=h264 --enable-pic
make -j4
sudo make install
```

...

but for whatever reason I always end up without vout_drm and vout_egl devices.
**Basically what i need is: hardware accelerated h265 decoding and output to vout_drm** (because `fbdev` lacks performance)

Can you spot what I am doing wrong?
Your help is very much appreciated. Thanks!

Best regards, Thomas

egnor, 2022-02-14:
It doesn't directly address this, but you've seen the "hello_drmprime" demo program, right? (https://github.com/jc-kynesim/hello_drmprime) Instead of using ffmpeg-integrated vout_drm, it does its own DRM/KMS display. Since I don't know your basic use case I can't say if that's helpful at all...

rstanislav, 2022-04-21:
Same problem here :( - latest rpi os bullseye, and no vout_drm.
Also during configure i get warning:

> WARNING: Option --enable-hwaccel=h264_v4l2request did not match anything

jc-kynesim, 2022-04-21:
What you are doing looks generally right to me - I admit that I normally compile on a "standard" (rather than lite) distribution so there might be a missing dependency in "apt build-dep ffmpeg" that I don't fall over.  I would expect

```
sudo apt-get build-dep ffmpeg 
pi-util/conf_native.sh --noshared
cd out/arm*
make -j8

```
to work and produce a staticly linked ffmpeg in the dest directory (name will vary depending on what you are building) as thats what I do.  Remove the --noshared to create dynamic libs but beware that you either need to use LD_LIBRARY_PATH or clean out the system libs if you are going to get consistent results.

jc-kynesim, 2022-04-21:
(You may find compile errors in egl_vout as it looks like I was over enthusiastic in removing spurious include files - patch RSN)

jc-kynesim, 2022-04-21:
> Same problem here :( - latest rpi os bullseye, and no vout_drm. Also during configure i get warning:
> 
> > WARNING: Option --enable-hwaccel=h264_v4l2request did not match anything

On a Pi H264 is only supported by stateful V4L2 so you want h264_v4l2m2m for that not hwaccel drm

spookyfirehorse, 2022-05-05:
hi to all

this is my config which enable vout drm and egl for 4.4 32 bit

./configure --prefix=/usr --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --incdir=/usr/include/arm-linux-gnueabihf  --arch=armhf --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2  --enable-neon --enable-v4l2-request --enable-libudev --enable-sand --enable-pocketsphinx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --enable-nonfree --enable-libfdk-aac --enable-version3 --disable-htmlpages --disable-manpages --disable-podpages --disable-txtpages --enable-epoxy --enable-libxcb --enable-libzimg --enable-vout-drm --enable-libv4l2 --enable-vout-egl --disable-static --enable-libsrt --extra-cflags=-I/usr/include/drm --enable-mmal --enable-rpi --cpu=cortex-a72 --arch=armhf 

when th configur is finish you see outdevs vout drm and egl

regards

spooky

closed 2025-06-05 by jc-kynesim