jc-kynesim/rpi-ffmpeg #32: Building on 64bits

lockywolf, 2022-01-25:
The official configure script seems to be expecting 32bits only, and is tied to rpi userspace (mmal).

Is it possible to build  rpi-ffmpeg on a 64bit system? (Slarm64 if that matters.)

jc-kynesim, 2022-01-25:
Try the pi-util/conf_native.sh script from dev/4.4/rpi_import_1 - that definitely works on PiOS 64 (I've just checked) and has options for mmal/no-mmal.  I probably haven't backported everything to 4.3.2.  The conf_pi* cross-compile scripts have undoubtedly rotted hard but were sort of hanging around as a hint I might want to fix them, but these days native compile is fast enough that it has not been any sort of priority.

lockywolf, 2022-01-25:
Hm.... can't say that upstream ffmpeg plays 4k h265 smoothly.... even if recoded with a -fastdecode profile.

But the code seem to have built fine, even though the prefixes are a bit messed up. I'll close.

closed 2022-01-25 by lockywolf

jc-kynesim, 2022-01-25:
If you have h/w (stateless V4L2) H265 decode and direct drm display then its smooth, otherwise not.