jc-kynesim/rpi-ffmpeg #33: FFmpeg has a memory leak in raspberry pi 4-based OSs

Minauras, 2022-01-27:
## Bug report
### Bug description
When playing videos with Kodi in either OSMC or Xbian on a Raspberry Pi CM4, a small amount of allocated heap memory is not freed when the video ends. These leaks build up and cause Kodi to crash after playing several videos.

Changing the allocated heap size with the cma option in either `/boot/config.txt` or `/boot/cmdline.txt` does not solve the problem, it only changes when it happens.

## Possible Fix
This behavior is not present on LibreELEC. On [this forum thread](https://forum.kodi.tv/showthread.php?tid=366631&pid=3082917#pid3082917), Team-Kodi member popcornmix mentions an [ffmpeg patch](https://github.com/LibreELEC/LibreELEC.tv/tree/libreelec-10.0/packages/multimedia/ffmpeg/patches/rpi) that is present on LibreELEC. Maybe it could be applied to other OSs.

### To Reproduce
Setup: Raspberry Pi CM4 on TOFU carrier board, with either OSMC or Xbian.

Either:
- Run a video playlist on loop and monitor the RAM or heap. Each time a video starts, the available memory drops. When the available memory reaches a critical state, Kodi crashes and restarts.
- Run a video and stop it, then do it again on loop. Monitor the RAM or heap. Each time the video starts, the available memory drops. When the available memory reaches a critical state, Kodi crashes and restarts.

### Debug log
[Here is a full log from a run with this bug.](https://gist.github.com/Minauras/bf71b945abc4acd3b299fbc2d68c9591)
The error is the following:
```
2022-01-21 14:34:33.951 T:1510     INFO <general>: ffmpeg[(nil)X]: Failed to alloc 12533760 from dma-heap(fd=176): 12 (Cannot allocate memory)
2022-01-21 14:34:33.951 T:1510    ERROR <general>: ffmpeg[(nil)X]: [hevc] v4l2_request_hevc_start_frame: Failed to get dst buffer
```

## Additional context
I am playing 4k24fps videos, I have not tried with other video formats.

I first reported this issue on the [Kodi forum](https://forum.kodi.tv/showthread.php?tid=366631&pid=3082917#pid3082917) and the [OSMC forum](https://discourse.osmc.tv/t/playling-playlist-dies-after-about-7-minutes-possible-memory-leak/92347).

I also raised this issue on the [XBMC github tracker](https://github.com/xbmc/xbmc/issues/20908).

### Environment
 - Operating system version/name: Either OSMC 2021.12-1 or Xbian 1.0 2021-11-22
 - Kodi version: 19.3.0
 - Hardware: Raspberry Pi CM4 on TOFU carrier board

jc-kynesim, 2022-01-27:
AFAIK LibreELEC take my dev/4.4/rpi_import_1 branch without further patching for the Pi, or at the very least without any patching that might fix CMA leaks. Pi use test/4.3.2/rpi_main on top of any debian patches which should be patched very close to 4.4.  The last leak I'm aware of I patched on 2021-10-07.

Minauras, 2022-01-27:
Thanks a lot for your reply. 🙂

Do you have any idea for a fix or a workaround for this issue?

jc-kynesim, 2022-01-27:
Ummm... No? Your problems are on distributions I don't use with a package I don't maintain - I've fixed my bit.  Sorry if that sounds callous, but I don't have control or particular understanding here.  For what limited consolation it might be I would expect my fixes to make it into your distribution at some point in the, hopefully not too distant, future - I do talk to the Kodi people regularly but I don't have any control over their distributions.
Sorry

closed 2022-01-27 by Minauras