jc-kynesim/rpi-ffmpeg #34: Cannot find hevc hw accelerated codec in c/c++ code

Consti10, 2022-01-29:
Hello,

I want to test the rpi hw decoder from c/c++ code (we need exact numbers on the decoding latency). Any ideas how to get the hw accelerated decoder (context) ? I've tried
`codec = avcodec_find_decoder_by_name("hevc_v4l2m2m");`

but this doesn't seem to work:
```
Found codec V4L2 mem2mem HEVC decoder wrapper
NO HW accel IN USE
[hevc_v4l2m2m @ 0x561e83bb20c0] Could not find a valid device
[hevc_v4l2m2m @ 0x561e83bb20c0] can't configure decoder
could not open codec
```

Here:
https://forums.raspberrypi.com/viewtopic.php?f=29&t=283301&sid=364ae48d72f4ef1048a5acc10df21be3&start=25#p1721203
you say:
hevc_v4l2m2m isn't expected to work - that uses the stateful v4l2 interface, the H265 decoder is stateless.

But how else do I find it in code then ?

Btw, we want to test if the rpi h265 decoder buffers any frames (I don't think stateless necessarily means the decoder doesn't buffer frames) but maybe you can also simply answer this question here.

Testing repo (I will probably extend it to a program similar to raspivid):
https://github.com/Consti10/ffmpeg_decoder/blob/3ba6f1a00b3f997afcc19da82a102421248eb226/ffmpeg_decoder.cpp

Consti10, 2022-01-29:
Well, I found a repo that looks to do exactly that -
https://github.com/jc-kynesim/hello_drmprime ;)

egnor, 2022-02-14:
Yep, that's the demo program you want to work from. As you probably figured out, the answer is to use the regular "hevc" codec, but enable the appropriate hardware decoding.

Hopefully you also got your question about frame buffering answered? (I'm not sure what you're asking there specifically.)

Consti10, 2022-02-27:
The question about frame buffering is simple - how can you get the HW decoder to not hold onto any decoded frame if the input h265 stream can be decoded wthout frame buffering.

I am pretty sure now that the 
`decoder_ctx->thread_count = 1;`

workaround works.

closed 2022-02-27 by Consti10