jc-kynesim/rpi-ffmpeg #35: Question: Is nb_layers in AVDRMFrameDescriptor ever != 1? (If so, what does it mean?)

egnor, 2022-02-14:
The [ffmpeg docs](https://ffmpeg.org/doxygen/3.4/structAVDRMFrameDescriptor.html#a228800d9ce195e776833e58ee307a6ec) say `nb_layers` describes the "number of layers in the frame".

The idea SEEMS to be that a video frame could be a composition of "layers", where each layer is an image, but I can't get a clear read on expectations. Is the idea that some codecs or filters might want to merge images into the final frame, but would produce them in this form so the merging can be done efficiently at display time? (This is DISTINCT from image component "planes", such as the Y/U/V planes in a planar YUV format. Each "layer" is a fully formed image.)

In [hwcontext_vulkan.c](https://github.com/jc-kynesim/rpi-ffmpeg/blob/release/4.3/rpi_main/libavutil/hwcontext_vulkan.c), `nb_layers` is set from `planes` which is set from `av_pix_fmt_count_planes(hwfc->sw_format)` which doesn't seem right per my understanding above that layers and planes are different things, but maybe I'm confused about something?

Most DRM-generating codecs (`v4l2m2m`, `hevc`, `kmsgrab`, `rkmpdec`) seem to just set `nb_layers` to 1.

One exception is [hwcontext_vaapi.c](https://github.com/jc-kynesim/rpi-ffmpeg/blob/release/4.3/rpi_main/libavutil/hwcontext_vaapi.c) which does seem to use or expect 2 layers in some cases, but seems to be confusing layers and planes in some way?? Maybe I don't understand the terminology?? The [vaapi code](https://github.com/intel/libva/blob/master/va/va_drmcommon.h#L93) seems to suggest that "layers" are only ever used to carry different image planes. It's all very confusing??

jc-kynesim, 2022-02-14:
Honestly I'm pretty much as confused as you are - not my structure. My understanding goes with your first guess. I don't know when they would be used - maybe if subtitles were coded as pics with the frame - or possibly for SVC style stuff?

egnor, 2022-02-14:
Fun threads that make things more confusing
- https://www.mail-archive.com/ffmpeg-devel@ffmpeg.org/msg83572.html
- https://github.com/intel/media-driver/issues/1210
- https://github.com/xbmc/xbmc/pull/16102

The layer option has been in since the DRM hardware context was first introduced
- https://www.mail-archive.com/ffmpeg-devel@ffmpeg.org/msg55566.html

There's a near-identical structure in the VA-API headers, not sure which came first
- https://github.com/intel/libva/commit/51e98b1224794a44ba097baa7a1b4e35c3596d0c

Note that the same Mark Thompson created both structures. I might just ask them out of the blue.

I was also guessing that it could be for something like late compositing of subtitles or something like that, BUT the only actual use I can find is for formats where for some reason Y and UV are represented as separate "layers" each with their own format, specifically NV12 gets represented as one R8 layer [sic! actually Y!] and one GR88 layer [sic! actually UV!]. Why anyone would represent the data that way as opposed to two planes I don't know.

jc-kynesim, 2022-02-15:
The vaapi will have come first, there is a bunch of idea copying from there.  The two plane R8 & RG8 stuff is GL and it actually makes a sort of sense if you don't have explicit YUV import but do have shaders that you can program to do the appropriate conversions (VLC will do that) but even there I wouldn't expect layers to be the right thing.

Consti10, 2022-02-27:
For h265 decoding, it looks to me as if (for the decoded frame(s) )
desc->nb_objects is always 1
and
desc->nb_layers is also always 1

jc-kynesim, 2022-02-28:
Using our h/w decoder that will always be the case; with my current understanding I can't see any reason why layers should be anything other than 1, but we could legitimately decode to 2 objects (luma in one and chroma in the other); that would be extra work but it would allow us to work around the fact that the parameters you require to describe a sand frame fit poorly into the std stride x height model.