jc-kynesim/rpi-ffmpeg #37: Is there any way of avoiding need for recent Mesa?

Fredrum, 2022-03-06:
Hello!

I'm getting near being able to post my new version of the Playstation remote client Chiaki which is leaning on ffmpeg to decode the streaming h264 video using m2m drmprime decoding.

So far I have needed using a Mesa newer than 21.0.1 (I'm currently using 21.3.1) but RPiOS Bullseye only comes with 20.3.5 which mean users will need to compile and install a newer version on their systems.

Do you know if there's any way at all of making this work with the Mesa that comes bundled with RPiOS Bullseye?

Cheers
Fred

jc-kynesim, 2022-03-07:
There is a current task to backport the needed patches but its stalled behind other work. I am also frustrated by this - I thought it had already happened but it turns out not.

Fredrum, 2022-03-07:
I see. That's great to be aware of though!
Do you know if there's an 'Issue' or a PR that I should keep an eye on for progress?

Fredrum, 2022-03-10:
Tomorrow I'll post on the RPi forums about the new Chiaki version I made. Thought I'd drop it here to as it's using the ffmepg work youv'e done :)

https://youtu.be/qDthHmD30us

https://github.com/Fredrum/chiaki/wiki/rpi02---branch-for-Bullseye,-64bit,-h265-and-kmsdrm

Cheers!