jc-kynesim/rpi-ffmpeg #39: av_frame_cropped_width(frame) Error on Non-Raspberry platforms

Fredrum, 2022-03-20:
Hello, I'm trying to build my project that uses drmprime m2m decode and render on the reaspberries, but now on things like Ubuntu.
Iv'e tried both a Raspi Ubuntu and my amd64 PC Ubuntu. Both times I got this error.
I'll dig some more to see what exactly is missing from where but I wanted to check if you knew about this?

For example is there a specific workaround you know?

```
/home/fred/user_data/chiaki-rpi/rpi/src/gui.cpp:885:10: error: ‘av_frame_cropped_width’ was not declared in this scope
  885 |   *a++ = av_frame_cropped_width(frame);
```

Cheers!
Fred

Fredrum, 2022-03-21:
Hi these errors come up with the default Ubuntu distributed 'AV' packages. I think I remember you saying that your patches were going to be added upstream somewhere so I was trying that out. Maybe that's not happened yet.
I 'll do some more investigations and write back here so you don't have to and get a bit more info.

jc-kynesim, 2022-03-21:
Indeed. My patches are not upstream (and I wouldn't hold your breath). av_frame_cropped_width is a short inline I wrote to simplify getting the width of a frame - by default ffmpeg just shifts the start of the data when cropping a frame, but the pi SAND formats can't be dealt with like that so cropping persists for longer.

Fredrum, 2022-03-21:
Ah ok I see! No worries!
I had a crazy idea that maybe I could 'cross use' packages between distros, like taking a RaspiOS AVCodec package and installing on say the equivalent architecture Ubuntu. But that's not a question for here! :)