jc-kynesim/rpi-ffmpeg #40: Could I extract 10bit luma data from AV_PIX_FMT_YUV420P10 HEVC?

Fredrum, 2022-04-03:
Hello! 

I have started a new project and am looking for a way to transmit 10(or 12)bit monochromatic depth camera data.
One way I thought would be 10bit hevc using avcodec.

I am sending from a Ubuntu-PC and receiving on a RPi4b.

On the sender side I am using AV_PIX_FMT_YUV420P10 at the moment.
This gives me this,

```
ffprobe video_dump.h265
Stream #0:0: Video: hevc (Rext), yuv420p10le(tv), 512x424, 30 fps, 30 tbr, 1200k tbn, 30 tbc
```

But I'm unsure how I can extract the Luma channel on the receiving end on the RPi.
I looked in the avcodec source code and found a function,

`int av_rpi_sand_to_planar_frame(AVFrame * const dst, const AVFrame * const src)`

But that crashes with a segfault when receiving the stream described above.

I am not fixed on needing to get this straight into OpenGL I'm hoping I'll be ok taking the luma pixels out to cpu memory as the resolution is not super high.

Cheers!

Fredrum, 2022-04-05:
Wait! I just realized I had forgotten to do
`tmpDst = av_frame_alloc();`
on the destination AVFrame! 

now the command goes though! :) 
`int av_rpi_sand_to_planar_frame(AVFrame * const dst, const AVFrame * const src)`

Exciting fingers crossed I will take a look at the data now! 

Cheers!

jc-kynesim, 2022-07-20:
Well the good news is that someone is looking into enabling GL SAND30 texture import - whether or not that will turn out to be successful or fast enough I don't know, but I wouldn't hold your breath.
Don't use the 4.3.1/drmprime_1 branch - its obsolete - current is test/4.3./rpi_main.
I've lost understanding of why you want to construct SAND30 frames (its a terrible format for most CPU driven ops)? Given current h/w capabilities the only reason I can think of would be if you want HDR output via DRM? I don't _think_ our GL allows HDR output currently???

jc-kynesim, 2022-07-20:
If you are outputting Dolby Video to the TV my understanding is that you need dynamic metadata attached to the frame - as it stands (unless you've done more mods) our DRM layer is capable of static HDR metadata, but not dynamic. (I've got non-Dolby HDR working via direct DRM)

jc-kynesim, 2022-07-20:
If you want example code that outputs HDR directly via DRM look at my drmu project (hello_drmu is a trivial video player).

docdude, 2022-07-20:
> If you want example code that outputs HDR directly via DRM look at my drmu project (hello_drmu is a trivial video player).

I already output HDR metadata through DRM and dovi standard and LLDV infoframes(with modded kernel DRMvc4).
Had to add code to the kernel to parse dolby VSDB, dynamic HDR, and dolby Vendor specific infoframes. 
```
static void drm_parse_cea_ext(struct drm_connector *connector,
			      const struct edid *edid)
{
	struct drm_display_info *info = &connector->display_info;
	const u8 *edid_ext;
	int i, start, end;

	edid_ext = drm_find_cea_extension(edid);
	if (!edid_ext)
		return;

	info->cea_rev = edid_ext[1];

	/* The existence of a CEA block should imply RGB support */
	info->color_formats = DRM_COLOR_FORMAT_RGB444;

	/* CTA DisplayID Data Block does not have byte #3 */
	if (edid_ext[0] == CEA_EXT) {
		if (edid_ext[3] & EDID_CEA_YCRCB444)
			info->color_formats |= DRM_COLOR_FORMAT_YCBCR444;
		if (edid_ext[3] & EDID_CEA_YCRCB422)
			info->color_formats |= DRM_COLOR_FORMAT_YCBCR422;
	}

	if (cea_db_offsets(edid_ext, &start, &end))
		return;

	for_each_cea_db(edid_ext, i, start, end) {
		const u8 *db = &edid_ext[i];

		if (cea_db_is_hdmi_vsdb(db))
			drm_parse_hdmi_vsdb_video(connector, db);
		if (cea_db_is_hdmi_forum_vsdb(db))
			drm_parse_hdmi_forum_vsdb(connector, db);
		if (cea_db_is_microsoft_vsdb(db))
			drm_parse_microsoft_vsdb(connector, db);
		if (cea_db_is_y420cmdb(db))
			drm_parse_y420cmdb_bitmap(connector, db);
		if (cea_db_is_vcdb(db))
			drm_parse_vcdb(connector, db);
		if (cea_db_is_hdmi_hdr_metadata_block(db))
			drm_parse_hdr_metadata_block(connector, db);
		if (cea_db_is_hdmi_hdr_dynamic_metadata_block(db))
			drm_parse_hdmi_hdr_dynamic_metadata_block(connector, db);
		if (cea_db_is_hdmi_hdr10_plus_vsdb(db))
			drm_parse_hdmi_hdr10_plus_vsdb(connector, db);
		if (cea_db_is_hdmi_dolby_vsdb(db))
			drm_parse_hdmi_dolby_vsdb(connector, db);
//		if (cea_db_is_hdmi_colorimetry_data_block(db))
//			drm_parse_colorimetry_data_block(connector, db);
	}
}
```

```
void ofxRPI4Window::updateHDR_Infoframe(hdmi_eotf eotf, int idx)
{
	bool ok;
	uint64_t blob_id = 0;	
	ofLog() << "DRM: Setting HDR infoframe";	

/*
	ok = drm_mode_get_property(device, connectorId,	DRM_MODE_OBJECT_CONNECTOR, "DOVI_OUTPUT_METADATA", &prop_id, &blob_id, &prop);
	if (!ok) {
		ofLogError() << "Unable to find DOVI_OUTPUT_METADATA";
[... 199 lines cut ...]
        float poly_coeffs[8][3];
        uint8_t mmr_order[8];
        float mmr_constant[8];
        float mmr_coeffs[8][3][7];
    } curves[3];

    struct dovi_nlq_t {
        uint8_t offset_depth; /* bit depth of offset value */
        uint16_t offset;
        uint64_t hdr_in_max;
        uint64_t dz_slope;
        uint64_t dz_threshold;
    } nlq[3];
} video_dovi_metadata_t;

void map_dovi_metadata(video_dovi_metadata_t *out, const AVDOVIMetadata *data);
void dovi_decode(AVFrame *tmp_frame, const video_dovi_metadata_t *data, int x, int y);

```

May be an exercise in futility but wanted to see if it was possible to bypass GL and use decoded dolby video pumped directly into DRM.  

Libplacebo proved GL works for dolby video using shaders.  It grabs the AVFrame pixel data and creates a texture which is then processed in shaders.  Slooow on pi4.   but works.  Want to do the same but thru DRM.

BTW I like what you did with drmu.

docdude, 2022-07-20:
> Having looked at your patches a bit harder - isn't it just a matter of dropping the dovi header into the dovi blob? I assume not as you wouldn't be asking me all this if it was that simple?

I wish.  But no.  The dovi blob merely sends an Vendor specific infoframe to set the dolby vision badge on the display.  An LLDV infoframe will set the dv badge but any dolby content still has to be decoded which is extremely slooow on the pi4 cpu and gpu for that matter with shaders.  A dolby STANDARD infoframe requires the content to have embedded rpu (vdr)color metadata inserted in the first 2 scanlines as described in the dolby patent available online.  With this VSIF and the correct AVIF, these dovi rpu pixels embedded, and correct even/odd pixel value format you get video, known as "rgb tunneling".  This is how we generate dolby vision patterns for our pattern generator to do dolby display calibration.  

>  If the latter then you/we are probably stuffed - you don't have enough time to do even the simplest transform on 4kp60 video on the CPU.

I think this will end up being the case, unfortunately.  

We can turn on the dolby vision badge sending the infoframes with the modded DRM code, but it still requires decoding the metadata in the AVFRAME to reshape the color of each frame in order to  correctly view dolby content.  
For pattern generation, using STANDARD dolby infoframe, we can embed the dovi color rpu, havent tried that on video though.
 Im thinking would still need the color reshaping plus it only works for 1080p patterns, and 8bit, not 4k AFAIK.  I've gotten 10bit patterns to work but you can see the dolby rpu metadata on the top2 scanlines since it is in 8bit.  
 Dolby documentation is very limited so most of this has been done by reverse engineering.

Ill keep u posted though.

popcornmix, 2022-07-20:
> but any dolby content still has to be decoded which is extremely slooow on the pi4 cpu

Can you explain roughly what this involves? Does it require processing per pixel for the decoded video frame?
If so roughly how many operations, and what is it doing this for?

docdude, 2022-07-20:
> Can you explain roughly what this involves? Does it require processing per pixel for the decoded video frame?
If so roughly how many operations, and what is it doing this for?

It is per pixel processing. You can see it requires multiple operations.  It is basically a rudimentary software dolby metadata decoder required to get the correct colors displayed from dolby content. 

This function pulls the dovi metadata from the AVFrame sidedata
```
void map_dovi_metadata(video_dovi_metadata_t *out,
                               const AVDOVIMetadata *data )
{	
    const AVDOVIRpuDataHeader *hdr = av_dovi_get_header( data );
    const AVDOVIDataMapping *vdm = av_dovi_get_mapping( data );
    const AVDOVIColorMetadata *color = av_dovi_get_color( data );
    out->bl_bit_depth = hdr->bl_bit_depth;
    out->el_bit_depth = hdr->el_bit_depth;
    out->coef_log2_denom = hdr->coef_log2_denom;
    out->nlq_method_idc = (enum dovi_nlq_method_t) vdm->nlq_method_idc;
    for( size_t i = 0; i < ARRAY_SIZE( out->nonlinear_offset ); i++ )
        out->nonlinear_offset[i] = av_q2d( color->ycc_to_rgb_offset[i] );
    for( size_t i = 0; i < 9; i++ ) {
		float *nonlinear_matrix = &out->nonlinear_matrix.m[0][0];

        nonlinear_matrix[i] = av_q2d( color->ycc_to_rgb_matrix[i] );
	

		float *linear_matrix = &out->linear_matrix.m[0][0];
        linear_matrix[i] = av_q2d( color->rgb_to_lms_matrix[i] );
	}
    out->source_min_pq = color->source_min_pq;
    out->source_max_pq = color->source_max_pq;
    for (int c = 0; c < 3; c++) {
        const AVDOVIReshapingCurve *csrc = &vdm->curves[c];
        struct dovi_reshape_t *cdst = &out->curves[c];
        cdst->num_pivots = csrc->num_pivots;
        for (int i = 0; i < csrc->num_pivots; i++) {
            const float scale = 1.0f / ((1 << hdr->bl_bit_depth) - 1);
            cdst->pivots[i] = scale * csrc->pivots[i];
        }
        for (int i = 0; i < csrc->num_pivots - 1; i++) {
            const float scale = 1.0f / (1 << hdr->coef_log2_denom);
            cdst->mapping_idc[i] = csrc->mapping_idc[i];
            switch (csrc->mapping_idc[i]) {
            case DOVI_RESHAPE_POLYNOMIAL:
                for (int k = 0; k < 3; k++) {
                    cdst->poly_coeffs[i][k] = (k <= csrc->poly_order[i])
                        ? scale * csrc->poly_coef[i][k]
                        : 0.0f;
                }
                break;
            case DOVI_RESHAPE_MMR:
                cdst->mmr_order[i] = csrc->mmr_order[i];
                cdst->mmr_constant[i] = scale * csrc->mmr_constant[i];
                for (int j = 0; j < csrc->mmr_order[i]; j++) {
                    for (int k = 0; k < 7; k++)
                        cdst->mmr_coeffs[i][j][k] = scale * csrc->mmr_coef[i][j][k];
                }
                break;
            }
        }
    }
 //   assert(sizeof(out->curves) == sizeof(vdm->curves));
    assert(sizeof(out->nlq)    == sizeof(vdm->nlq));
  //  memcpy(out->curves, vdm->curves, sizeof(out->curves));
    memcpy(out->nlq,    vdm->nlq,    sizeof(out->nlq));
    for( size_t i = 0; i < ARRAY_SIZE( out->curves ); i++)
        assert( out->curves[i].num_pivots <= ARRAY_SIZE( out->curves[i].pivots ));
}
```

This takes the dovi metadata and does the color reshaping.
[... 342 lines cut ...]
		color[i] = pow(MAX(0, Em2 - c1) / (c2 - c3 * Em2), 1 / m1);
	}
}

void OETF(float color[3]) 
{
	// Common constants for SMPTE ST.2084 (PQ)
    static const double m1 = 2610.0 / 4096 * 1./4,
						m2 = 2523.0 / 4096 * 128,
						c1 = 3424.0 / 4096,
						c2 = 2413.0 / 4096 * 32,
						c3 = 2392.0 / 4096 * 32;
    for (int i=0; i < 3; i++) {
		if (color[i] > 0) {
			double Ym1 = pow(color[i], m1);
			color[i] = pow((c1 + c2 * Ym1) / (1 + c3 * Ym1), m2);
		} else {
			color[i] = 0;
		}
	}
}	
```

Again this is code mostly ported from libplacebo, I took the shaders it creates for dolby processing and converted to C. 
Ran the code against an shader debugger using the libplacebo shaders for dovi and the pixel values come out the same for different dovi profiles i tested.

jc-kynesim, 2022-07-21:
Thanks for that

Am I right that if you output a Dolby standard header & encode the appropriate data in the picture you don't need to manipulate the rest of the data or is some other processing still required?  Sorry if I'm failing to understand what you've said - you know a lot more about this than me and things that are obvious to you are still new to me.

Looks like Dolby are deliberately making it hard for people to do easy implementations. - I can't see any way that we could setup our h/w to get all the bottom bits right as is required by that patent.  Do you happen to have timings for how long it takes Pi4 shaders to process a frame (ignoring all other frame import/export related sloth)?

[Having said the above I wonder if we could in fact characterise the HVS conversions s.t. we could manipulate the input data to get known bottom bits in the output at the expense of some minor fidelity in the top 3 lines - we could afford quite a lot of processing to get 3 lines right if it saved us processing the whole frame]

docdude, 2022-07-25:
Sorry for the late response. 

> Am I right that if you output a Dolby standard header & encode the appropriate data in the picture you don't need to manipulate the rest of the data or is some other processing still required?

The metadata embedded in the NAL unit of the video stream tells the dolby decoder how to reshape the picture. 
The input metadata bitstream contains:
```

Parsing RPU file...
{
  "dovi_profile": 8,
  "header": {
    "rpu_nal_prefix": 25,
    "rpu_type": 2,
    "rpu_format": 18,
    "vdr_rpu_profile": 1,
    "vdr_rpu_level": 0,
    "vdr_seq_info_present_flag": true,
    "chroma_resampling_explicit_filter_flag": false,
    "coefficient_data_type": 0,
    "coefficient_log2_denom": 23,
    "vdr_rpu_normalized_idc": 1,
    "bl_video_full_range_flag": false,
    "bl_bit_depth_minus8": 2,
    "el_bit_depth_minus8": 2,
    "vdr_bit_depth_minus_8": 4,
    "spatial_resampling_filter_flag": false,
    "reserved_zero_3bits": 0,
    "el_spatial_resampling_filter_flag": false,
    "disable_residual_flag": true,
    "vdr_dm_metadata_present_flag": true,
    "use_prev_vdr_rpu_flag": false,
    "prev_vdr_rpu_id": 0,
    "vdr_rpu_id": 0,
    "mapping_color_space": 0,
    "mapping_chroma_format_idc": 0,
    "num_pivots_minus_2": [
      0,
      0,
      0
    ],
    "pred_pivot_value": [
      [
        0,
        1023
      ],
      [
        0,
        1023
      ],
      [
        0,
        1023
      ]
    ],
    "nlq_method_idc": null,
    "nlq_num_pivots_minus2": null,
    "nlq_pred_pivot_value": null,
    "num_x_partitions_minus1": 0,
    "num_y_partitions_minus1": 0
  },
  "rpu_data_mapping": {
    "mapping_idc": [
      [
        0
      ],
      [
        0
      ],
      [
[... 534 lines cut ...]
[154] out_color = _main_18_2();
[155] }
vk->CreateDescriptorSetLayout(vk->dev, &dinfo, PL_VK_ALLOC, &pass_vk->dsLayout)
vk->CreatePipelineLayout(vk->dev, &linfo, PL_VK_ALLOC, &pass_vk->pipeLayout)
vk_compile_glsl(gpu, tmp, GLSL_SHADER_VERTEX, params->vertex_shader, &vert)
shaderc compile status 'success' (0 errors, 0 warnings)
**Spent 352.064 ms translating SPIR-V (slow!)**
vk_compile_glsl(gpu, tmp, GLSL_SHADER_FRAGMENT, params->glsl_shader, &frag)
shaderc compile status 'success' (0 errors, 0 warnings)
**Spent 1519.914 ms translating SPIR-V (slow!)**
```
The shader processing latency is in bold.  
**Spent 352.064 ms translating SPIR-V (slow!)**
**Spent 1519.914 ms translating SPIR-V (slow!)**

The dolby reshaping process in short is this:
- Import the Luma and 2 chroma pixel values
- Normalize the values to [0,1]
- Polynomial reshaping, i.e. polynomial coeffs used on Luma, MMR coeffs used on chroma
- Apply non-linear matrix(ycc_to_rgb coeffs, BT2020 or BT709) and ycc_rgb_offsets  **//this can be done in CSC**
- Apply PQ 
- Apply RGB_to_LMS matrix  **//this can be done in CSC**
- Apply inverse PQ

I don't think the pi4 is up to this unfortunately.

jc-kynesim, 2022-07-26:
Many thanks for the comprehensive answer.  I have to agree that it doesn't look in any way likely that a Pi is going to be up for that.  I admit I was hoping that it was going to be TV-side processing rather than source-side.  Ah well - at least when I say "we can't do Dolby Video" I am better informed about why.  I guess this way maximises the number of licenses they get to sell.

We'd already noticed the lack of perfection in YUV translation when doing other HDR work but I don't think there's anything we can do about it - on the other hand for non-conformance grade work they are close enough.

Thanks again

docdude, 2022-07-28:
Your welcome.  
I was wondering if HVS has a 12bit or 16bit format.  Mesa has exposed 16bit. I have a utility ive written to write to HVS directly from userspace and have not been able to find a format other than what is already coded in kernel.

jc-kynesim, 2022-07-31:
I think the only >8 bit formats in HVS are 10-bit XRGB packed in 32-bit words and 10-bit YUV 4:2:0 in SAND30. But I'm not an expert on the h/w

popcornmix, 2022-08-01:
> I think the only >8 bit formats in HVS are 10-bit XRGB packed in 32-bit words and 10-bit YUV 4:2:0 in SAND30. But I'm not an expert on the h/w

I don't see any other high bit depth formats in the spec.

docdude, 2022-08-01:
Thanks again.

naushir, 2022-12-15:
@docdude sorry for stirring up this old issue, but I wonder if you can provide some more information on the general topic of Dolby Vision?  This is mainly for my understanding more than anything else!

1) I understand that in order to support DV over HDMI 1.4, we need to "tunnel" the 12-bit DV  image + metadata over  RGB 8-bit video signal.  Your above code essentially does this.  Can we signal DV content natively in HDMI 2.0 without the need to do this tunnelling? If yes, does the DV metadata get sent through an av infoframe stream?

2) Could you encode a HEVC stream directly in DV ICtCp format?  If so, would there be any per-pixel operations required (like you described above) to send this over HDMI 2.0 in DV format? If yes, can you provide a brief idea of what's required?

3) Does DV require ICtCp encoding, or can we use the more common YCbCr encoding?  Perhaps this is based on levels/profiles used?

closed 2025-06-05 by jc-kynesim