jc-kynesim/rpi-ffmpeg #43: Keeping buffers generated by h264_v4l2m2m leads to errors

egnor, 2022-04-17:
Hello kind person,

In my application, I use libav* to decode video files, but in some cases I will keep the decoded frames around without releasing them. Specifically I want to keep the dma-buf ("DRM prime") buffers from various frames and be able to show them on screen. Obviously this is limited by CMA but that's adjustable up to the size of physical memory...

But if I "borrow" enough frames that way, h264_v4l2m2m is unable to keep decoding, I get these log messages

```
07:00:49.854     24u T [libav] h264_v4l2m2m[0xb2127b30] V4L2 poll capture empty
07:00:49.854     21u T [libav] h264_v4l2m2m[0xb2127b30] V4L2 poll capture empty
07:00:49.854     20u T [libav] mov,mp4,m4a,3gp,3g2,mj2[0xb2100d50] stream 0, sample 135, dts 5360000
07:00:49.854     44u T [libav] h264_v4l2m2m[0xb2127b30] V4L2 poll capture empty
07:00:49.854     25u T [libav] h264_v4l2m2m[0xb2127b30] In PTS=70144, DTS=68608, track=136, n=136
07:00:49.854     23u T [libav] h264_v4l2m2m[0xb2127b30] V4L2 poll output ret=0, timeout=0, events=0x104, revents=0
07:00:49.854     19u T [libav] h264_v4l2m2m[0xb2127b30] V4L2 poll capture empty

   (the last two lines repeat a lot)

07:00:49.866     31u T [libav] h264_v4l2m2m[0xb2127b30] V4L2 poll output ret=0, timeout=0, events=0x104, revents=0
07:00:49.866     18u T [libav] h264_v4l2m2m[0xb2127b30] V4L2 poll capture empty
07:00:49.866     18u E [libav] h264_v4l2m2m[0xb2127b30] Unexpectedly large retry count: 256
07:00:49.866    223u E [loader] Decode packet (h264_v4l2m2m): Input/output error
```

My theory is that h264_v4l2m2m only keeps a finite number of buffers in a pool, and once I have "borrowed" enough of them without releasing the reference, it no longer has any to cycle back into the decoder to receive new frames. Is that... accurate? Is there any way around this, by somehow messing with the parameters of that buffer pool, or "refilling" it with new buffers, or something?

Interestingly, the hevc decoder (which of course uses a different kernel interface) doesn't _seem_ to have this problem, does that make sense?

-- egnor

egnor, 2022-04-17:
Actually, I spoke too soon about hevc, it seems to have a similar problem, though it can give up a few more frames before running into the wall:

```
07:04:23.476 784090u D [libav] hevc[0xb21472e0] nal_unit_type: 0(TRAIL_N), nuh_layer_id: 0, temporal_id: 0
07:04:23.476    144u D [libav] hevc[0xb21472e0] v4l2_req_frame_alloc: avctx=0xb21472e0 size=680 data=0x2ea4f60
07:04:23.476     55u D [libav] hevc[0xb21472e0] Output frame with POC 29.
07:04:23.478   1924u D [libav] hevc[0xb21088f0] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
07:04:23.478    145u D [libav] hevc[0xb21088f0] v4l2_req_frame_alloc: avctx=0xb21088f0 size=680 data=0xaee76ea0
07:04:23.478     37u D [libav] hevc[0xb21088f0] Output frame with POC 30.
07:04:23.478     52u E [libav] hevc[0xb21088f0] create_dst_bufs: Failed to create V4L2 buffer
07:04:23.479     93u E [libav] hevc[0xb21088f0] v4l2_request_hevc_start_frame: Failed to get dst buffer
07:04:23.479     56u E [libav] hevc[0xb21088f0] create_dst_bufs: Failed to create V4L2 buffer
07:04:23.479     23u E [libav] hevc[0xb21088f0] v4l2_request_hevc_end_frame: Failed to get dst buffer
07:04:23.479     17u E [libav] hevc[0xb21088f0] hardware accelerator failed to decode picture
07:04:23.479     21u D [libav] v4l2_req_frame_free: avctx=0xb21088f0 data=0xaee76ea0
07:04:23.479     65u T [libav] mov,mp4,m4a,3gp,3g2,mj2[0xb2100d50] stream 0, sample 33, dts 1240000
07:04:23.479    114u T [libav] mov,mp4,m4a,3gp,3g2,mj2[0xb2100d50] stream 0, sample 34, dts 1280000
```

I don't _think_ the "Failed to create V4L2 buffer" is an actual out-of-memory from the kernel, but I'm not 100% sure of that.

egnor, 2022-04-17:
For example, one hack might be, on getting an AVFrame from the codec with an AVDRMObjectDescriptor inside, ...

- copy the descriptor, including the fd, into my private world
- allocate a new DRM buffer of appropriate size
- _replace_ the fd field in the AVDRMObjectDescriptor with the allocated buffer
- free the AVFrame, thus returning my newly allocated buffer to the pool

Would that let me "steal" buffers? Or is that a losing proposition because there are other data structures storing the list of "known" buffers? (There's also the issue of figuring out how to allocate that buffer...)

egnor, 2022-04-18:
After doing some research, the hack described above will NOT work, because there is indeed a lot of V4L2-specific context. The V4L2 code in libavcodec uses `VIDIOC_REQBUFS` to get a fixed pile of buffers, referred to by index, and hands those out. The DRM file descriptors are exported from those buffers. Messing with the file descriptors by themselves would just confuse things.

Maybe what I want is just to bump the `num_capture_buffers` option to a large number (defaults to 20, looks like?) so there will be a lot more "headroom". I fear that a large number would lead to a ton of wasted memory. I'm not clear on the lifecycle of the actual RAM attached to a V4L2 buffer... ideally the driver would use a memory-attached buffer in the recycle queue before allocating new memory for a fresh buffer... but I bet drivers just use the recycled buffers in FIFO order?? I'm not sure I'm even making sense here.

egnor, 2022-04-19:
I sent a message to the shark infested waters of libav-user, we'll see if anyone bites there!
http://ffmpeg.org/pipermail/libav-user/2022-April/013058.html

jc-kynesim, 2022-04-21:
You are definitely in interesting waters here.  For H264 (stateful V4L2) the buffers are allocated by V4L2 which sizes them based on what it knows about frame dimensions, format etc.  The fact they are allocated by V4L2 as part of the Q also limits their number to (I think) 32 - on reflection I'm not sure why I think that so I might be wrong.
My H265 V4L2 stateless decode allocs its own dmabufs so in theory should be more flexible, but in practice it was easier to bind them to Q entries and offhand I'm not sure how easy it would be to extract them (its been a while since I wrote that now) but I'm guessing not easy.
In your application you know what you are doing but many applications are very incautious about allocating buffers and having a limited pool that stalls on alloc works much better than letting them just run us out of CMA...

egnor, 2022-04-21:
Huh, how do you alloc an individual dmabuf? Does the stateless interface provide a way to do that "a la carte"?

Anyway, here's what I ended up doing for now:

I track how many buffers I have "checked out" on a per decoder basis, by creating an atomic counter that I increment when receiving a frame and decrement when calling av_frame_free / av_frame_unref. (I never make additional frame references otherwise things would get complicated! Would be nice if I could get that stat from the source instead of tracking it myself, but eh.)

When I want to use a frame, I look at that counter. If it's below a certain threshold (currently 15) I go ahead and "steal" the frame as long as I need it. If it's above that threshold, I make a deep copy of the image data and then release the frame.

That way I get the zero copy benefit of frame stealing much of the time, but when I happen to be retaining a lot of frames I make sure there's still some buffers left for the decoder to use.

It would still be better to be able to dynamically alloc a new buffer for the decoder to use instead of memcpy-ing several megabytes of frame, of course. But it seems like that would be quite nontrivial, and for now I'm still in business.

Closing this issue because I think we're all on the same page (the same buffer?) and I'm hardly going to ask for a major architectural change! (Honestly if I want to do better, probably writing my own V4L2 client is the quickest path? Easy enough for H.264, not at all easy for HEVC...)

closed 2022-04-21 by egnor

jc-kynesim, 2022-04-21:
> Huh, how do you alloc an individual dmabuf? Does the stateless interface provide a way to do that "a la carte"?

Generic dmabufs can be got from the `/dev/dma_heap/linux,cma` device (look at libavcodec/v4l2_req_dmabufs.c) and then just dropped into V4L2; you still need slots in the V4L2 Q but, in theory, the slots don't have to be filled with the same dmabuf each time.  You can do this with stateful too and if I'd felt the need to do a major rewrite on it then I would have done it that way.
 
> Anyway, here's what I ended up doing for now:
>
> I track how many buffers I have "checked out" on a per decoder basis, by creating an atomic counter that I increment when receiving a frame and [...]

Seems like a plausible scheme to me

> Closing this issue because I think we're all on the same page (the same buffer?) and I'm hardly going to ask for a major architectural change! (Honestly if I want to do better, probably writing my own V4L2 client is the quickest path? Easy enough for H.264, not at all easy for HEVC...)

Stateless is has more bits than stateful (though I would argue that stateful is more fiddly to get right), the framework that I dropped on stateless decode is probably more complex than truly required but I'd already written it and I knew it worked well with the Pi decode.

Milkysunshine, 2026-09-21:
I believe I hit the same underlying issue with Firefox as the DRM PRIME consumer, on Qualcomm MSM8953/APQ8053 using the current test/9.0/main branch.

Firefox retains decoded DRM PRIME/DMABUF surfaces for presentation. With the default num_capture_buffers=20, H.264 decoding repeatedly stalled after only a few frames. The most useful observation was that when Firefox released one retained surface, FFmpeg immediately requeued the corresponding V4L2 capture buffer and the hardware decoder immediately produced another frame.

I changed num_capture_buffers in libavcodec/v4l2_m2m_dec.c from 20 to 32. With the same Firefox build and only the FFmpeg library rebuilt, the capture starvation disappeared and sustained hardware-decoded playback began working, including 1080p60 H.264.

So at least in this case, increasing the capture-buffer pool was enough to accommodate an application retaining DRM PRIME frames.

Full notes, revision/build details and exact patch:
https://gist.github.com/Milkysunshine/cdc40638348393967b4e2f13b623cefb

My Firefox build also contains an experimental avcodec_send_packet(EAGAIN) handling change, but that change alone did not fix the 20-buffer stall. Playback became stable when the capture-buffer count was increased.

jc-kynesim, 2026-09-21:
The problem as far as I'm concerned with that is that is that you need a lot of memory for a lot of buffers and a number of the Pis that this is meant to run on have a very limited amount of that so I'm unkeen to change that default.

Now there are two ways that I can see to deal with this given that you have control over the calling code.

1) You could just set the "capture_frames" option in the options dict passed in to avcodec_open2. This would work right now.
2) I would happily fix the code s.t. it uses the avctx->extra_hw_frames parameter and you could set that. I'm slightly surprised that I haven't done that already.

(2) is, I think, the approved ffmpeg way of dealing with h/w devices that have fixed pools like we have here, and where the caller has a need to hold on to some buffers for some period rather than simply returning them to ffmpeg asap.

Milkysunshine, 2026-09-21:
That parameter would be much more helpful than throwing random numbers in the compile to see where Firefox finally gives up on me. 

My testing was done on a Lenovo ThinkSmart (originally Teams version) with 2GB of ram. I realize there are Pis with less, but this isn't really a work horse. That is the main reason I chose this ffmpeg to work with.  

I tested by building rpi-ffmpeg exactly as sourced, and changing only these in Firefox, and it worked:

```diff
diff --git a/dom/media/platforms/ffmpeg/FFmpegVideoDecoder.cpp b/dom/media/platforms/ffmpeg/FFmpegVideoDecoder.cpp
index 16c2a415d6..82c1f37199 100644
--- a/dom/media/platforms/ffmpeg/FFmpegVideoDecoder.cpp
+++ b/dom/media/platforms/ffmpeg/FFmpegVideoDecoder.cpp
@@ -821,7 +821,25 @@ MediaResult FFmpegVideoDecoder<LIBAV_VER>::InitV4L2Decoder() {
     return ret;
   }

-  if (mLib->avcodec_open2(mCodecContext, codec, nullptr) < 0) {
+  AVDictionary* v4l2Options = nullptr;
+
+  int optionResult =
+      mLib->av_dict_set(&v4l2Options, "num_capture_buffers", "32", 0);
+
+  if (optionResult < 0) {
+    mLib->av_dict_free(&v4l2Options);
+    FFMPEG_LOG("  Failed to set V4L2 num_capture_buffers=32");
+    return NS_ERROR_DOM_MEDIA_FATAL_ERR;
+  }
+
+  FFMPEG_LOG("  Requesting 32 V4L2 capture buffers");
+
+  int v4l2OpenResult =
+      mLib->avcodec_open2(mCodecContext, codec, &v4l2Options);
+
+  mLib->av_dict_free(&v4l2Options);
+
+  if (v4l2OpenResult < 0) {
     FFMPEG_LOG("  Couldn't initialise V4L2 decoder");
     return NS_ERROR_DOM_MEDIA_FATAL_ERR;
   }

diff --git a/toolkit/moz.configure b/toolkit/moz.configure
index 2a0b994468..3e265bd685 100644
--- a/toolkit/moz.configure
+++ b/toolkit/moz.configure
@@ -945,6 +945,8 @@ def eme_choices(target, wmf):
         return ("widevine",)
     if target.os in ("OSX"):
         return ("widevine",)
+    if target.kernel == "Linux" and target.cpu == "aarch64":
+        return ("widevine",)

 # Widevine is enabled by default in desktop browser builds.
```

 Much cleaner fix... thanks for the suggestion.

Thank you for the project!