jc-kynesim/rpi-ffmpeg #44: build error related to dav1d 1.0.0 update on dev/4.4/rpi_import_1

graysky2, 2022-05-08:
When trying to build latest kodi on arm64 using your patched ffmpeg from https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.4/rpi_import_1 I am getting build errors like this:
```
src/libavcodec/libdav1d.c:150:52: error: ‘DAV1D_MAX_TILE_THREADS’ undeclared (first use in this function); did you mean ‘DAV1D_MAX_THREADS’?
  150 |                      : FFMIN(floor(sqrt(threads)), DAV1D_MAX_TILE_THREADS);
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~
src/libavutil/common.h:105:28: note: in definition of macro ‘FFMIN’
  105 | #define FFMIN(a,b) ((a) > (b) ? (b) : (a))
```

[Complete ](https://gist.githubusercontent.com/graysky2/70c50c14075fd158114f0590cdf533e4/raw/dca67d5c8f1705690bb4f31132acbc1a3e2a24df/build.log)build log.

I am thinking that it is something in your patch.  Building the same commit of kodi using vanilla ffmpeg 4.4.1 on x86_64, builds without error.

jc-kynesim, 2022-05-08:
Dunno. I haven't intentionally touched dav1d in any way and a `git diff --stat n4.4 --` doesn't reveal any hints as to a tweaked file in that area. Maybe its simply that my branch is based off n4.4 (i.e. the 4.4 root) rather than the current 4.4 tip and dav1d has fixed something between then and now?

nyanmisaka, 2022-05-08:
ffmpeg 4.4 doesn't compliant with dav1d 1.0 API. You need this patch or downgrade to 0.92.

https://github.com/FFmpeg/FFmpeg/commit/e204846ec16c1ab34c7f3a681734cf5190433018

graysky2, 2022-05-08:
I cannot explain why but the build works now.  I created a new branch, rebased it against kodi/master manually correcting conflicts successfully.  I find zero differences, yet it compiles successsfully.  I will close this, thank you for looking @jc-kynesim

```
% git diff gs-gbm_nexus-v3 gs-gbm_nexus-v4
% echo $?
0
```

graysky2, 2022-05-09:
@nyanmisaka - are you certain?  When I look at the [ffmpeg tarball](https://github.com/xbmc/FFmpeg/tags) kodi requires [4.4.1-Nexus-Alpha1](https://github.com/xbmc/FFmpeg/releases/tag/4.4.1-Nexus-Alpha1), most of that patch is already applied.  Just the last two lines are not included.

nyanmisaka, 2022-05-09:
4.4 != 4.4.1

graysky2, 2022-05-10:
@jc-kynesim - there is definitely something in your code -- or in the patch I generated perhaps incorrectly from it -- causing this.  If I build kodi using the commit from @popcornmix (https://github.com/popcornmix/xbmc/commit/49471532e1af659d074791c27a7e231052071951), it completes without error.  If however I overwrite his patch ([tools/depends/target/ffmpeg/0001-rpi-Add-hevc-acceleration.patch](https://github.com/popcornmix/xbmc/commit/49471532e1af659d074791c27a7e231052071951#diff-0c9a2f56da783b77321761df65f5f42af55c027862b778eb4f6e0193fc51ffcf)) with your latest code from dev/4.4/rpi_import_1, I get the build error.  I generated the patch which I just copied over his like this:
```
git checkout dev/4.4/rpi_import_1
git fetch xbmc
git diff xbmc/release/4.4-kodi > tools/depends/target/ffmpeg/0001-rpi-Add-hevc-acceleration.patch
```

> Maybe its simply that my branch is based off n4.4 (i.e. the 4.4 root) rather than the current 4.4 tip and dav1d has fixed something between then and now?

I am unsure how to test this...

@popcornmix - Did you generate `0001-rpi-Add-hevc-acceleration.patch` in your gbm branch as I did above?  Any thoughts?

jc-kynesim, 2022-05-11:
I'd try creating the patch from n4.4 as the base (as that is where the branch is based) - not xbmc/release/4.4-kodi - or merge with xbmc/release/4.4-kodi first maybe (or possibly from whatever that is using as a root these days)?  Getting a "correct" kodi tree from the multiple places that patches are assembled from can be a little tricky.

graysky2, 2022-05-12:
I tried from n4.4 and that gave an error-free build.  Thanks for the suggestion.

closed 2022-05-12 by graysky2

popcornmix, 2022-05-12:
@graysky2 script I use for updating kodi and LE ffmpeg patches.
```
 git fetch xbmc
 S="-o LogLevel=ERROR"

 scp $S dom@buildbot:~/projects/LibreELEC.tv/packages/multimedia/ffmpeg/package.mk /tmp
 HASH=$(grep -A4 "RPi)" /tmp/package.mk | grep "PKG_VERSION=" | head -1 | cut -c17- | tr -d '"' | cut -c-40)
 if [ "$HASH" == "" ]; then
  HASH=$(grep "PKG_VERSION=" /tmp/package.mk | cut -c13- | tr -d '"')
 fi
 echo HASH:$HASH
 ssh $S dom@buildbot "rm -f ~/projects/LibreELEC.tv/packages/multimedia/ffmpeg/patches/v4l2-drmprime/*"
 ssh $S dom@buildbot "rm -f ~/projects/LibreELEC.tv/packages/multimedia/ffmpeg/patches/v4l2-request/*"
 ssh $S dom@buildbot "rm -f ~/projects/LibreELEC.tv/packages/multimedia/ffmpeg/patches/libreelec/ffmpeg-001-libreelec.patch"
 git diff-index $HASH --binary > /tmp/0001-rpi-Add-hevc-acceleration.patch
 scp $S /tmp/0001-rpi-Add-hevc-acceleration.patch dom@buildbot:~/projects/LibreELEC.tv/packages/multimedia/ffmpeg/patches/rpi/ffmpeg-001-rpi.patch

 HASH=$(grep "VERSION=" ~/bullseye/home/pi/projects/kodi/kodi_gbm/tools/depends/target/ffmpeg/FFMPEG-VERSION | cut -c9- | tr -d '"')
 echo HASH:$HASH
 git diff-index $HASH --binary > ~/bullseye/home/pi/projects/kodi/kodi_gbm/tools/depends/target/ffmpeg/0001-rpi-Add-hevc-acceleration.patch
```

graysky2, 2022-05-12:
@popcornmix - Thanks for sharing that

graysky2, 2022-05-12:
@popcornmix - Which branch from jc-kynesim are you using for:
* Matrix
* Nexus (assuming [dev/4.4/rpi_import_1](https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.4/rpi_import_1))