jc-kynesim/rpi-ffmpeg #45: Trying to make this work on Ubuntu - has it been tried?

Fredrum, 2022-05-22:
Hello!

I'm now trying to make your rpi ffmpeg work on Ubuntu 64bit on an RPi4b!
Im getting some fresh error messages here and there but I just tried running this and the output looks strange to me,

`./ffmpeg -no_cvt_hw -hwaccel drm -vcodec hevc -i ~/Videos/jellyfish-50-mbps-hd-h264.mkv -f null -`

It looks like its identifying the stream as hevc when it should be h264?

Has this ffmpeg version been tested yet on Ubuntu? And do you have any advice for making it work?

Cheers!
Fred

Output:
ffmpeg version pi/4.3.3/rpi_12-3-g4ac70f56f6 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.2.0-7ubuntu2)
  configuration: --enable-shared --disable-static --enable-sand --enable-v4l2-request --enable-libdrm --enable-libudev --enable-opengl
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
[hevc @ 0xaaaadfc21650] Invalid NAL unit size in extradata.
[matroska,webm @ 0xaaaadfc1f0b0] Failed to open codec in avformat_find_stream_info
[NULL @ 0xaaaadfc21650] Invalid NAL unit size in extradata.
[NULL @ 0xaaaadfc21650] Invalid NAL unit 0, skipping.
    Last message repeated 1 times
[NULL @ 0xaaaadfc21650] Invalid NAL unit 38, skipping.
[NULL @ 0xaaaadfc21650] Invalid NAL unit 0, skipping.
[NULL @ 0xaaaadfc21650] Invalid NAL unit 29, skipping.
    Last message repeated 2 times
[NULL @ 0xaaaadfc21650] Invalid NAL unit size (39854 > 2524).
[hevc @ 0xaaaadfc21650] Invalid NAL unit size in extradata.
Input #0, matroska,webm, from '/home/fred/Videos/jellyfish-50-mbps-hd-h264.mkv':
  Metadata:
    encoder         : libebml v1.2.0 + libmatroska v1.1.0
    creation_time   : 2016-02-06T04:00:55.000000Z
  Duration: 00:00:30.03, start: 0.000000, bitrate: 49901 kb/s
    Stream #0:0(eng): Video: hevc, none, 1920x1080, SAR 1:1 DAR 16:9, 29.97 fps, 29.97 tbr, 1k tbn, 1k tbc (default)
[hevc @ 0xaaaadfc3bfd0] Invalid NAL unit size in extradata.
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> wrapped_avframe (native))
Error while opening decoder for input stream #0:0 : Invalid data found when processing input

Fredrum, 2022-05-22:
Actually re-opening...did anyone make this work on Ubuntu-Rpi4b?
If so, any advice?

jc-kynesim, 2022-05-22:
Are you asking if my tree compiles on rpi ubuntu (The 'this' in the issue title is somewhat ambiguous)?  I haven't tried.  If you can characterise what fails I'll have a quick look.

Fredrum, 2022-05-22:
HI can change the title when I know more what I'm asking! :)

My Setup:
Linux pi-ubuntu 5.13.0-1026-raspi #28-Ubuntu SMP PREEMPT Mon Apr 25 15:22:19 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
OS: Ubuntu 21.10 aarch64

Actually after thinking a bit clearer this morning it seems like I'm either missing or the configure script is not finding EGL.

It builds fine using this config line,
`./configure --enable-shared --disable-static --enable-sand --enable-v4l2-request --enable-libdrm --enable-libudev --enable-opengl --enable-vout-egl  --enable-vout-drm`

But the out devices are missing your egl one,
```
Enabled outdevs:
alsa                    oss                     v4l2
fbdev                   sdl2                    vout_drm
opengl                  sndio                   xv
```

I did installed these,
`sudo apt install libglfw3-dev libgles2-mesa-dev`

Ubuntu comes with,
`OpenGL version string: 2.1 Mesa 21.2.6`

And the raw ffmpeg seems to work as this runs ok:
`./ffmpeg -no_cvt_hw -hwaccel drm -vcodec h264 -i ~/Videos/jellyfish-50-mbps-hd-h264.mkv -f null -`

but this does not, since I have an EGL problem:
./ffmpeg -no_cvt_hw -hwaccel drm -vcodec h264 -i ~/Videos/jellyfish-50-mbps-hd-h264.mkv -f vout_egl /dev/null

Probably this does not actually have to do with ffmpeg as such it feels more like I've got something missning somewhere.
So feel free to leave me to it.  :)

jc-kynesim, 2022-05-23:
I'd go though the headers included in libavcodec/egl_vout.c and see what is missing on your system.  If none are missing then I've got a config problem. If I had to guess a missing package I'd look at libepoxy-dev (I'm guessing the name) as that's not a default package for most things. (Used by me 'cos it was used by the original author of the code I copied.)

Fredrum, 2022-05-24:
I looked in egl_vout.c and compared the includes to the ones in the enabled drm_vout.c and the suspect differences were,
```

#include <epoxy/gl.h>
#include <epoxy/egl.h>
...
#include <X11/Xlib.h>
#include <X11/Xutil.h>
```
so I ran apt on libepoxy-dev but that was already installed and latest version.

One other actually was
`#include <stdatomic.h>`
but I'm assuming that comes with gnu c/c++.
libx11-dev was also already included as I mentioned earlier.

Soo...not sure?

Fredrum, 2022-05-25:
Thanks for the tip I think I used something like apt-file a while ago but then forgot about it again.
libxv-dev was already installed on my system. The problem is that I don't know if I can find an output or record of why vout_egl got rejected? I also can't find where to read the rule that decides if its ok to build it?

Xvlib.h was found here,
/usr/include/X11/extensions/Xvlib.h

[config.h.gz](https://github.com/jc-kynesim/rpi-ffmpeg/files/8767214/config.h.gz)
[config.log.gz](https://github.com/jc-kynesim/rpi-ffmpeg/files/8767215/config.log.gz)

jc-kynesim, 2022-05-25:
Add --enable-epoxy to your config. It is arguably a bug you need it (and I'll have a go at fixing it) as it should be a dependency not something you have to have set but I think that will fix your problem.  Tip: when your custom config fails have a look at what I'm using in pi-util/conf_native.sh as that is what I use to configure stuff.  ffmpeg configure is something I'm still confused by.

Fredrum, 2022-05-25:
Nice one thanks I'll try these steps out tonight!  I think it makes sense to fix in the config like you said although for me I was going to write an install script for 'my' users to make it super easy. I'll let you know how it works out with --enable-epoxy or your new config.

jc-kynesim, 2022-05-25:
dev/4.4 is used by Kodi and generally maintained in sync with test/4.3.2/rpi_main which is the Pi (debian) distribution. In this case I've just pushed to 4.4 so I can get your feedback before I commit the changes to the other branch.

Fredrum, 2022-05-26:
Alright that got us a step further so that's great!
But I now get this error message,
`[vout_egl @ 0xaaab12410620] Frame (format=0) not DRM_PRiME`
I'll add the whole chunk at the end here.
This was doing both vout_egl on the desktop and vout_drm in the tty. Same message.
I checked ffmpeg,
```
ldd ffmpeg | grep drm
libdrm.so.2 => /lib/aarch64-linux-gnu/libdrm.so.2 (0x0000ffffa14ec000)
```

```
ldd ffmpeg | grep dri
ldd ffmpeg | grep gbm
```
both returned empty but I'd have to check on my RaspiOS Bullseye to see if I should expect any of those.

I might have to let this simmer a bit and taking a look at libnraries attached on the RaspberryOS version and see if I can find some differences.

jc-kynesim, 2022-05-27:
What was your command line?  I see no evidence of accelerated decode there and the vout_egl/drm devices are very stupid such that they only accept DRM_PRIME frames (and the error says they aren't getting that).
I'd be using something like: ` ./ffmpeg -no_cvt_hw -c:v h264_v4l2m2m -i ~/rpi/streams/jellyfish-3-mbps-hd-h264.mkv -f vout_egl /dev/null`
Maybe `-v trace` would shed light on what's being tried (or maybe hide it in too much debug)

doesnotexist, 2022-05-27:
I successfully built dev/4.4/rpi_import_1 on the rpi ubuntu 22.04 (jammy) after installing libxv-dev libglfw3-dev libgles2-mesa-dev libdrm-dev libudev-dev   .  Although, libx11-dev was already installed pkg-config complained about no xlib until I installed the libxv-dev like you mentioned above.  For configuring I did
`pi-util/conf_native.sh --noshared`
and then
`make -j8 -C out/arm64-jammy-4.4-static-rel`

The problem I have now is finding a command line that works to confirm that the hardware accelerated decode is working.
I tried `.out/arm64-jammy-4.4-static-rel/ffmpeg -no_cvt_hw -c:v h264_v4l2m2m -i ~/jellyfish-3-mbps-hd-h264.mkv vout_egl /dev/null`

but that errors with
`[NULL @ 0xaaab29c26ee0] Unable to find a suitable output format for 'vout_egl'
vout_egl: Invalid argument

Fredrum, 2022-05-27:
I'll take a look at this again on Sat evening as I'm tied up until then.
Iv'e been using these command lines,
`./ffmpeg -no_cvt_hw -hwaccel drm -vcodec h264 -i ~/Videos/jellyfish-50-mbps-hd-h264.mkv -f vout_egl /dev/null`
that I must have gotten from earlier testing.
Is `-no_cvt_hw -hwaccel drm` not telling it to do hw accell?

@doesnotexist 
Is it because yo uforgot the **-f** in-front of 'vout_egl /dev/null'  ?
I see youv'e got -j8 on your make what hardware are you running on?

doesnotexist, 2022-05-27:
I built it on the raspberry pi 4  running ubuntu jammy, I probably shouldn't have done -j4 but I was just copy/pasting what was in the pi-utils/BUILD.txt

@Fredrum Thanks for catching that missing -f, that was it!  So it appears to work now with this:
`.out/arm64-jammy-4.4-static-rel/ffmpeg -no_cvt_hw -c:v h264_v4l2m2m -i ~/jellyfish-3-mbps-hd-h264.mkv -f vout_egl /dev/null`

Using vout_egl, displaying the decoded images still requires quite a bit of CPU from what I can tell. All 4 cores go to 100%.  This doens't seem to be the case if I do it with vout_drm from the terminal.

Fredrum, 2022-05-27:
@jc-kynesim 
Arr... yes you're right I see now that in my old notes/posts I was using `h264_v4l2m2m `but at some point for this round of testing that became `h264`!
Ha that might mean it's already actually working fine!

Fredrum, 2022-05-27:
Sorry for the multiple posts.
@doesnotexist  
It shouldn't take all that cpu with EGL. I get around 18-25% cpu in resource mon (which I think adds up all threads) and definitely nowhere near 100%.

Maybe try adding `-hwaccel drm` to your command line?

doesnotexist, 2022-05-27:
@Fredrum  `-hwaccel drm` has no effect for me when using vout_egl from gnome/wayland.  My guess is that the cpu spike has happens outside of ffmpeg's direct control and more to do with the compositing and scaling (full screen in vout_drm I'm assuming does no scaling).  Or perhaps some extra copying of the decoded frames is happening in route to being displayed within a wayland window.  Watching bpytop while the video decodes, I never see the ffmpeg process uses more than 11% of cpu.

jc-kynesim, 2022-05-28:
`-hwaccel drm` is what you need for h265 (V4L2 stateless), `-c:v h264_v4l2m2m` is what you need for H264 (V4L2 stateful). Note that with a simple `-f vout_egl` (or drm) ffmpeg will decode as fast as it can, dropping frames if display can't keep up and so you will get inflated (vs. real-time playback) CPU usage. If you want something a little closer to reality add` -show_all 1` before the` -f` which means that all decoded frames should be displayed (i.e. you get 1 frame per vsync), but if your display rate doesn't match the video rate you will get fast/slow playback.
Just for clarity I'll repeat my standard disclaimers: vout_drm/egl are test/example output devices, come as-is, and have known issues, any real-world usage is at your own risk.

Fredrum, 2022-05-29:
Now I'm trying with,
`./ffmpeg -no_cvt_hw -vcodec h264_v4l2m2m -i ~/Videos/jellyfish-50-mbps-hd-h264.mkv -f vout_egl /dev/null`
also tried
`./ffmpeg -no_cvt_hw -hwaccel drm -vcodec h264_v4l2m2m -i ~/Videos/jellyfish-50-mbps-hd-h264.mkv -f vout_egl /dev/null`

It's not complaining about a DRM Frame anymore but instead I get this,
(It seems to do some work but no image or window appears)

```
Input #0, matroska,webm, from '/home/fred/Videos/jellyfish-50-mbps-hd-h264.mkv':
  Metadata:
    encoder         : libebml v1.2.0 + libmatroska v1.1.0
    creation_time   : 2016-02-06T04:00:55.000000Z
  Duration: 00:00:30.03, start: 0.000000, bitrate: 49901 kb/s
    Stream #0:0(eng): Video: h264 (High), yuv420p(tv, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], 29.97 fps, 29.97 tbr, 1k tbn, 59.94 tbc (default)
[h264_v4l2m2m @ 0xaaab051443d0] level=42
[h264_v4l2m2m @ 0xaaab051443d0] Using device /dev/video10
[h264_v4l2m2m @ 0xaaab051443d0] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0xaaab051443d0] requesting formats: output=H264 capture=YU12
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_v4l2m2m) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
[h264_v4l2m2m @ 0xaaab051443d0] V4L2 capture poll unexpected timeout: events=0x106
    Last message repeated 1 times
```

I run with -v trace and you can see the full output here,

https://pastebin.com/FMa2Nw4a

jc-kynesim, 2022-05-29:
Well the unexpected timeout thing is almost certainly 'cos the V4L2 stateful driver in the Ubuntu kernel (that is what you are using isn't it?) is somewhat behind the PiOS one. There was a bug in our driver some time ago where it didn't deal with the startup sequence correctly, and I'm pretty sure that what you are seeing here is that.

Fredrum, 2022-05-29:
Ok yes I was starting to think that maybe there might be some kernel differences.
Do you know what files/edits are involved or would it be better to ask that on the Raspberry OS kernel git issues area?

jc-kynesim, 2022-05-29:
I could make educated guesses but you are probably better off asking. I don't know how Ubuntu & Pi deal with patches.

jc-kynesim, 2022-06-02:
linux,cma is the device that I use for getting dmabufs (I don't know if system needs better perms). It is possible that under unbuntu it has different default perms to PiOS where it has group video rw.

jc-kynesim, 2022-06-02:
Maybe what you actually need is "sudo adduser fred video"?

Fredrum, 2022-06-02:
Thanks!  I will try that method instead on my next run! 
Now that's its clear that I can get this to work on RPi Ubuntu I will make my 'ffmpeg' install script for the Chiaki repo and then flash a fresh Ubuntu SD card to check all the steps.