jc-kynesim/rpi-ffmpeg #52: configure fails with "vout_egl requires epoxy"

roger-, 2022-10-29:
Trying to build the [dev/5.1.2/rpi_import_1](https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/5.1.2/rpi_import_1) branch per instructions but it fails at this part of the script:

```
 --enable-vout-egl\
```

Seems to work if I remove that option (not sure what the consequences are).

jc-kynesim, 2022-10-30:
What distribution are you building on? On PiOS I would expect apt-get build-dep ffmpeg to have installed the relevant libs.
There only consequences are that the ffmpeg doesn't have the vout_egl device that I use for testing drm_prime output via egl.

roger-, 2022-10-30:
I'm building on a Debian 11 (bullseye) Docker image.

Resulting binary seems okay (has other issues though), but would be nice to have a cleaner solution.

closed 2025-06-05 by jc-kynesim