jc-kynesim/rpi-ffmpeg #53: Non-playable stream with warning: "Failed to set number of B-frames: Invalid argument"

roger-, 2022-10-30:
I've compiled the [5.1.2/rpi_import_1](https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/5.1.2/rpi_import_1) branch and I'm trying to convert an MJPEG stream to h264 and publish it to rtsp-simple-server using this command:

```
ffmpeg -hide_banner -avoid_negative_ts make_zero -fflags nobuffer -flags low_delay -strict experimental -fflags +genpts+discardcorrupt -use_wallclock_as_timestamps 1 -f mjpeg -re -i http://xxx  -an -c:v h264_v4l2m2m -pix_fmt yuv420p -g 10 -r 10 -f rtsp -rtsp_transport tcp rtsp://localhost:8554/mystream
```

It seems to run, but the resulting stream never loads. I also see this message:

```
[h264_v4l2m2m @ 0x557b57e9c0] Failed to set number of B-frames: Invalid argument
[h264_v4l2m2m @ 0x557b57e9c0] Failed to set gop size: Invalid argument
```

The "gop" issue was mentioned [here](https://github.com/jc-kynesim/rpi-ffmpeg/issues/49) already and the "B-frames" issues was also mentioned [here](https://forums.raspberrypi.com/viewtopic.php?t=331832). Using a build of ffmpeg from   [here](https://johnvansickle.com/ffmpeg/) everything works okay and I don't get the "B-frames" message.

Here's a full ffmpeg log if that helps:

```
[mjpeg @ 0x557b579360] unable to decode APP fields: Invalid data found when processing input                                                                                                                           Input #0, mjpeg, from 'http://xxx':                                                                                                                                      Duration: N/A, start: 1667151868.389465, bitrate: N/A                                                                                                                                                                  Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 960x720, 25 tbr, 1200k tbn                                                                                                              Stream mapping:                                                                                                                                                                                                          Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (h264_v4l2m2m))                                                                                                                                                          Press [q] to stop, [?] for help                                                                                                                                                                                        [mjpeg @ 0x557b579cd0] unable to decode APP fields: Invalid data found when processing input                                                                                                                           [swscaler @ 0x557b5dd3d0] [swscaler @ 0x557b5eabc0] deprecated pixel format used, make sure you did set range correctly                                                                                                [swscaler @ 0x557b5dd3d0] [swscaler @ 0x557b621440] deprecated pixel format used, make sure you did set range correctly                                                                                                [swscaler @ 0x557b5dd3d0] [swscaler @ 0x557b657cc0] deprecated pixel format used, make sure you did set range correctly                                                                                                [swscaler @ 0x557b5dd3d0] [swscaler @ 0x557b68e540] deprecated pixel format used, make sure you did set range correctly          
...
[h264_v4l2m2m @ 0x557b57e9c0]  <<< v4l2_encode_init: fmt=0/-1
[h264_v4l2m2m @ 0x557b57e9c0] Using device /dev/video11
[h264_v4l2m2m @ 0x557b57e9c0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x557b57e9c0] requesting formats: output=YU12 capture=H264
[h264_v4l2m2m @ 0x557b57e9c0] Failed to set number of B-frames: Invalid argument
[h264_v4l2m2m @ 0x557b57e9c0] Failed to set gop size: Invalid argument
Output #0, rtsp, to 'rtsp://localhost:8554/mystream':
  Metadata:
    encoder         : Lavf59.27.100
  Stream #0:0: Video: h264, yuv420p(tv, bt470bg/unknown/unknown, progressive), 960x720, q=2-31, 200 kb/s, 10 fps, 90k tbn
    Metadata:
      encoder         : Lavc59.37.100 h264_v4l2m2m
[mjpeg @ 0x557b579cd0] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 2 times
[mjpeg @ 0x557b579cd0] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
[mjpeg @ 0x557b579cd0] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 3 times
```

And ffprobe gives:

```
[rtsp @ 000002175d1bdf00] Missing PPS in sprop-parameter-sets, ignoring
[h264 @ 000002175d1c3780] non-existing PPS 0 referenced
    Last message repeated 1 times
[h264 @ 000002175d1c3780] decode_slice_header error
[h264 @ 000002175d1c3780] no frame!
```

domsson, 2022-11-01:
> Using a build of ffmpeg from [here](https://johnvansickle.com/ffmpeg/) everything works okay and I don't get the "B-frames" message.

Does this mean you got hardware accelerated video encoding and streaming on the Pi via RTSP to work with that version of ffmpeg? If so, what is your reason to still attempt this with the jc-kynesim fork?

roger-, 2022-11-01:
@domsson yes, that other build seems to work with hardware **encoding**. Unfortunately not with **decoding**, which seems to be [broken with recent kernels](https://github.com/blakeblackshear/frigate/issues/3780).

(jc-kynesim's fork is supposed to include a fix, but I haven't gotten it working.)

spookyfirehorse, 2023-05-08:
and the -nobuffer option  on ffmpeg  5.1.2  32 bit is not working

spookyfirehorse, 2023-05-08:
and last

bookworm ffmpeg 5.1.2

bookworm 32 bit

git clone -b test/5.1.2/main https://github.com/jc-kynesim/rpi-ffmpeg.git

./configure --prefix=/usr --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --incdir=/usr/include/arm-linux-gnueabihf --extra-libs=-latomic --extra-cflags=-I/usr/include/libdrm --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --enable-nonfree --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libpulse --enable-nonfree --enable-libfdk-aac --enable-version3 --disable-htmlpages --disable-manpages --disable-podpages --disable-txtpages --enable-vulkan --arch=armv7l --disable-mmal --enable-libv4l2 --enable-vout-drm --enable-vout-egl --enable-v4l2-request --enable-libudev --enable-neon --enable-sand --enable-pocketsphinx --disable-static --enable-opencl --enable-libsrt --enable-librabbitmq --enable-librav1e

make -j4

sudo make install