jc-kynesim/rpi-ffmpeg #59: Requirements for hevc_v4l2m2m?

alucryd, 2022-11-16:
Hi there, I recently discovered this fantastic fork which finally fixed my h264_v4l2m2m woes with mpv on the pi 4. However hevc_v4l2m2m still doesn't work for me, what could I be missing?

I'm using https://github.com/jc-kynesim/rpi-ffmpeg/commit/8bba3c74aba3ab00eb4634c25573bad1a3c8b91c right now, and tried a variety of mpv releases from 0.32 to the newly released 0.35. Everything is built using a crosstool toolchain, and tested on official raspbian images, both 32 and 64 bit. Do I need to use a branch based on 5.1.2, if so which one?

Happy to provide more information.

jc-kynesim, 2022-11-16:
h265/hevc accel is not provided by v4l2 m2m. The H265 kernel driver uses the stateless V4L2 interface (rather than m2m which is stateful), so you need a command line of the form ffmepg -hwaccel drm -i myhevc.mkv...

alucryd, 2022-11-16:
That explains it, thanks for the explanation. So I guess on the mpv side that would translate to something like `--vo=gpu --gpu-context=drm --drm-connector=1.HDMI-A-1`? Will give that a try tomorrow.

harryqt, 2025-02-26:
Hi, any updates on this?

jc-kynesim, 2025-02-26:
What sort of update were you hoping for? The h/w doesn't support a stateful h265 decode therefore the ffmpeg stateful driver doesn't work and is never going to.