jc-kynesim/rpi-ffmpeg #61: Hardware accelerated decode in mpv

EmperorPenguin18, 2022-12-06:
Hello, I've been trying to get hardware accelerated playback with H264 (v4l2m2m) and HEVC (drm_prime) to work on mpv. The closest I've gotten is tracked [here](https://github.com/mpv-player/mpv/issues/10773). I was using the ffmpeg-rpi package created by the Arch Linux ARM folks which if I understand correctly uses some code from this repository. The conclusion at the end of that thread is that ffmpeg is outputting to an image format that mpv doesn't support. I was wondering why this is the case and if the situation could be remedied somehow? This issue may also be related to https://github.com/jc-kynesim/rpi-ffmpeg/issues/60

jc-kynesim, 2022-12-06:
H264 should produce "normal" format images i.e. NV12 or I420, H265 produces SAND8 (for 8-bit) or SAND30 (for 10-bit). This is what the h/w does and can't be changed. Obviously you can convert them in s/w but that takes too long for 4k.
The SAND8/30 formats can be consumed directly if you are outputting to DRM, and EGL knows about SAND8 and will shortly know about SAND30 (in HoT mesa as well as Pi backports).

vanfanel, 2022-12-10:
This interests me a lot, too

EmperorPenguin18, 2022-12-11:
> H264 should produce "normal" format images i.e. NV12 or I420, H265 produces SAND8 (for 8-bit) or SAND30 (for 10-bit). This is what the h/w does and can't be changed. Obviously you can convert them [...]

When you say EGL knows about it, are you referring to https://gitlab.freedesktop.org/mesa/drm/-/blob/main/include/drm/drm_fourcc.h#L964 till https://gitlab.freedesktop.org/mesa/drm/-/blob/main/include/drm/drm_fourcc.h#L1010 ?

jc-kynesim, 2022-12-12:
I mean 'knows about' in the sense that you can pass it to eglCreateImageKHR and it "just works"

closed 2023-01-04 by EmperorPenguin18