jc-kynesim/rpi-ffmpeg #63: HW accel decoding from dev/4.4/rpi_import_1 does not work with ffmpeg-4.4.3

graysky2, 2022-12-18:
Hi John.  When I build ffmpeg-4.4.3 for RPi4 using your https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.4/rpi_import_1 branch (`git diff n4.4`), the resulting executable does not work.

```
Assertion !avctx->hwaccel failed at libavcodec/pthread_frame.c:249
```
However, if I use the identical patch built against ffmpeg-4.4.2, everything works as expected.  Any ideas?

graysky2, 2022-12-19:
When I build kodi with ffmpeg-4.4.3 patched as above, it seems to die on the same key error when trying to playback HEVC sample:
```
2022-12-19 16:53:05.842 T:117158   debug <general>: OnPlayMedia /mnt/media/4k-jellyfish-120-mbps-4k-uhd-hevc-10bit.mkv
2022-12-19 16:53:05.860 T:117158   debug <general>: CPlayerCoreFactory::GetPlayers(/mnt/media/4k-jellyfish-120-mbps-4k-uhd-hevc-10bit.mkv)
2022-12-19 16:53:05.861 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: system rules
2022-12-19 16:53:05.861 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: matches rule: system rules
2022-12-19 16:53:05.861 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: mms/udp
2022-12-19 16:53:05.861 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: lastfm/shout
2022-12-19 16:53:05.862 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: rtmp
2022-12-19 16:53:05.862 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: rtsp
2022-12-19 16:53:05.862 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: streams
2022-12-19 16:53:05.862 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: dvd
2022-12-19 16:53:05.862 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: discimage
2022-12-19 16:53:05.863 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: sdp/asf
2022-12-19 16:53:05.863 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: nsv
2022-12-19 16:53:05.863 T:117158   debug <general>: CPlayerSelectionRule::GetPlayers: considering rule: radio
2022-12-19 16:53:05.863 T:117158   debug <general>: CPlayerCoreFactory::GetPlayers: matched 0 rules with players
2022-12-19 16:53:05.863 T:117158   debug <general>: CPlayerCoreFactory::GetPlayers: adding videodefaultplayer (VideoPlayer)
2022-12-19 16:53:05.864 T:117158   debug <general>: CPlayerCoreFactory::GetPlayers: for video=true, audio=false
2022-12-19 16:53:05.864 T:117158   debug <general>: CPlayerCoreFactory::GetPlayers: for video=true, audio=true
2022-12-19 16:53:05.864 T:117158   debug <general>: CPlayerCoreFactory::GetPlayers: added 1 players
2022-12-19 16:53:05.879 T:117158   debug <general>: Radio UECP (RDS) Processor - new CDVDRadioRDSData
2022-12-19 16:53:05.879 T:117158   debug <general>: Audio ID3 tag processor - new CVideoPlayerAudioID3
2022-12-19 16:53:05.879 T:117158    info <general>: VideoPlayer::OpenFile: /mnt/media/4k-jellyfish-120-mbps-4k-uhd-hevc-10bit.mkv
2022-12-19 16:53:05.880 T:117307   debug <general>: Thread VideoPlayer start, auto delete: false
2022-12-19 16:53:05.880 T:117158   debug <general>: OnPlayBackStarted: CApplication::OnPlayBackStarted
2022-12-19 16:53:05.881 T:117308   debug <general>: Thread BackgroundLoader start, auto delete: false
2022-12-19 16:53:05.882 T:117307    info <general>: Creating InputStream
2022-12-19 16:53:05.895 T:117281   debug <general>: Loading settings for /mnt/media/4k-jellyfish-120-mbps-4k-uhd-hevc-10bit.mkv
2022-12-19 16:53:05.900 T:117307   debug <general>: ScanForExternalSubtitles: Searching for subtitles...
2022-12-19 16:53:05.900 T:117307   debug <general>: ScanForExternalSubtitles: END (total time: 0 ms)
2022-12-19 16:53:05.900 T:117307    info <general>: Creating Demuxer
2022-12-19 16:53:05.901 T:117158   debug <general>: CVideoGUIInfo::InitCurrentItem(/mnt/media/4k-jellyfish-120-mbps-4k-uhd-hevc-10bit.mkv)
2022-12-19 16:53:05.902 T:117158   debug <general>: CPlayerGUIInfo::InitCurrentItem(/mnt/media/4k-jellyfish-120-mbps-4k-uhd-hevc-10bit.mkv)
2022-12-19 16:53:05.907 T:117307   debug <general>: Open - probing detected format [matroska,webm]
2022-12-19 16:53:05.908 T:117307   debug <general>: Open - avformat_find_stream_info starting
2022-12-19 16:53:05.918 T:117308   debug <general>: Thread BackgroundLoader 545563995456 terminating
2022-12-19 16:53:05.979 T:117307   debug <general>: Open - av_find_stream_info finished
2022-12-19 16:53:05.980 T:117307    info <general>: ffmpeg[0x55bcfa5a70]: Input #0, matroska,webm, from '/mnt/media/4k-jellyfish-120-mbps-4k-uhd-hevc-10bit.mkv':
2022-12-19 16:53:05.980 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:   Metadata:
2022-12-19 16:53:05.980 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:     COMPATIBLE_BRANDS: iso4hvc1iso6
2022-12-19 16:53:05.980 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:     MAJOR_BRAND     : iso4
2022-12-19 16:53:05.980 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:     MINOR_VERSION   : 1
2022-12-19 16:53:05.981 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:     ENCODER         : Lavf56.3.100
2022-12-19 16:53:05.981 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:   Duration: 00:00:30.10, start: 0.067000, bitrate: 119845 kb/s
2022-12-19 16:53:05.981 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:   Stream #0:0(und): Video: hevc (Main 10), yuv420p10le(tv), 3840x2160 [SAR 1:1 DAR 16:9], 29.97 fps, 29.97 tbr, 1k tbn, 29.97 tbc (default)
2022-12-19 16:53:05.981 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:     Metadata:
2022-12-19 16:53:05.981 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:       CREATION_TIME   : 2016-02-05 13:00:09
2022-12-19 16:53:05.981 T:117307    info <general>: ffmpeg[0x55bcfa5a70]:       LANGUAGE        : und
[... cut ...]
2022-12-19 16:53:05.994 T:117317   debug <general>: CVideoPlayerVideo - Stillframe left, switching to normal playback
2022-12-19 16:53:06.002 T:117158   debug <general>: ------ Window Init (DialogBusy.xml) ------
2022-12-19 16:53:06.004 T:117307   debug <general>: CVideoPlayer::CheckContinuity - wrapback :2, prev:133000.000000, curr:0.000000, diff:-133000.000000
2022-12-19 16:53:06.088 T:117309   debug <general>: ffmpeg[0x0]: Assertion !avctx->hwaccel failed at src/libavcodec/pthread_frame.c:249
```

jc-kynesim, 2023-01-05:
Sorry about the delay - pretty sure that is going to be the same assert I've fixed in 5.1.x - I'll get to that soon (hopefully without needing another branch)

graysky2, 2023-01-05:
Thanks John.  I will watch for a push to your https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.4/rpi_import_1

closed 2023-01-09 by graysky2