jc-kynesim/rpi-ffmpeg #66: Identify this fork when using libavutil

EmperorPenguin18, 2023-01-05:
I need to use preprocessors to identify this fork, or else AV_PIX_FMT_RPI4_8 will not be recognized if someone isn't compiling against this fork. Is there a #define that already exists that I missed? If not could one be added?

jc-kynesim, 2023-01-05:
Fair comment. I'll try to add a define to avutil/avconfig.h (which is included pixfmt.h) by  that depends on whether sand is configured in the build. (I'm sure that this is the wrong thing to do for upstream purposes but seems right in this context)

jc-kynesim, 2023-01-05:
Thought about it a bit more and added "#define AVUTIL_HAVE_PIX_FMT_SAND 1" unconditionally to pixfmt.h
Now on test/4.3.5/rpi_main. It'll percolate to other branches as I update stuff.
I'll add #defines for other features if you need them

EmperorPenguin18, 2023-01-06:
Could you add this to a branch based on a newer version of ffmpeg?

jeeb, 2023-01-07:
I think there has been some miscommunication since one can just check for the pix_fmt in either a configure script check or otherwise (see f.ex. https://code.videolan.org/videolan/libplacebo/-/merge_requests/273/diffs?commit_id=3f50487ccc8478f5d4e96a1c2c9436d06ba00d0f#8d1aaa9f698b380c799d54c1b163e10d4a07318f_1008_1023 ). With upstreamed changes the library's bumped-at-addition version is also commonly utilized as a check (a la https://github.com/mpv-player/mpv/commit/1d15a5a059064216d97775181506b7f8432eb30d#diff-30d8f6be6320feeacf686be94f48c70869b52630e01ea625f0f15adc0d57c3e4R600 ).

closed 2023-01-10 by EmperorPenguin18