jc-kynesim/rpi-ffmpeg #67: streaming with h264'_v4l2m2m  error h264: non-existing PPS 0 referenced

spookyfirehorse, 2023-01-05:
hi and hello !!!

ffmpeg 4.3.5

i stream with this comand

ffmpeg   -hide_banner  -async 1  \
  -f alsa  -ac 1 -i hw:CARD=Device,DEV=0  -f v4l2 -input_format h264   \
  -re   -i /dev/video0 -c:v copy  -pix_fmt yuv420p    \
  -c:a libopus  -b:a 32k  -application lowdelay   \
  -f rtsp -rtsp_transport tcp  rtsp://localhost:8557/mystream

and recieving with

mpv rtsp://spook:password@lucy:8557/mystream

stream is running but when recieving with mpv

error  

h264: non-existing PPS 0 referenced
[ffmpeg/video] h264: decode_slice_header error
[ffmpeg/video] h264: no frame!

i ask rtsp-simple-server and he answerd with

this was not generating SPS/PPS with FFmpeg 4.3.5, while it's generating SPS/PPS with FFmpeg 5.1.1.
Anyway, the server is only responsible for receiving data from FFmpeg, it doesn't influence how the data is generated. If problem persists, it's a FFmpeg issue.

thank you for your big great work

spookyfirehorse, 2023-05-03:
on 4.3.6 same error 
is there no solution ?

any help is welcome

new ffmpeg comand is

ffmpeg -hwaccel drm -hwaccel_output_format drm_prime -fflags +nobuffer   -use_wallclock_as_timestamps 1 -avioflags direct -flags low_delay -probesize 320 -analyzeduration 32  -hide_banner    \
  -f alsa   -i plughw:CARD=S3,DEV=0  -f v4l2 -input_format h264  -itsoffset 0.90  -f v4l2  -i /dev/video0  -vcodec h264_v4l2m2m -g 45 -b:v 1500k -acodec libfdk_aac  -b:a 96k  -map 1:0 -map 0:0    \
   -threads 4  -f rtsp -rtsp_transport tcp  rtsp://localhost:8554/mystream.sdp

mpv  rtsp://localhost:8554/mystream.sdp

h264: non-existing PPS 0 referenced
[ffmpeg/video] h264: decode_slice_header error
[ffmpeg/video] h264: no frame!

searching for any solution

jc-kynesim, 2023-05-03:
On the information given - dunno. Try outputting to a file rather than than rtsp so we can see what is coming out of ffmpeg?

jc-kynesim, 2023-05-03:
Fair enough. Assuming you can compile ffmpeg - try this patch:

```
diff --git a/libavformat/rtspenc.c b/libavformat/rtspenc.c
index 97e3ef6da3..0ab8875d24 100644
--- a/libavformat/rtspenc.c
+++ b/libavformat/rtspenc.c
@@ -250,6 +250,6 @@ AVOutputFormat ff_rtsp_muxer = {
     .write_header      = rtsp_write_header,
     .write_packet      = rtsp_write_packet,
     .write_trailer     = rtsp_write_close,
-    .flags             = AVFMT_NOFILE | AVFMT_GLOBALHEADER,
+    .flags             = AVFMT_NOFILE,
     .priv_class        = &rtsp_muxer_class,
 };
```
I don't guarantee it'll fix anything but it fixed my test setup (which isn't in any way the same as yours expect that it tried to use an rtsp transport).

jc-kynesim, 2023-05-03:
If that doesn't work I have some substantially more complicated things to try.

spookyfirehorse, 2023-05-03:
yes i think so its a long time  bug history and sorry for my bad english

you update 2 month before ffmpeg i think 4.1 or 4.2 with this versions are allready ok with rtsp

maybe it helps

spookyfirehorse, 2023-05-07:
hi and hello !

I hope there comes a new version that works!
as of this writing i use copy option and -f v4l2 -input_format h264 and it works !
the new ffmpeg 5.1.2 update 2 days ago doesn't work at all for rtsp h264_v4l2m2m non monutonus output or something 

no luck for now with diff --git a/libavformat/rtspenc.c b/libavformat/rtspenc.c

but i use copy so no problem

And thanks for everything

great work

spookyfirehorse, 2023-05-07:
so quick 4.3.6 on bullseye running
thank you so much

spookyfirehorse, 2023-05-07:
sorry its working on 32 bit 5.1.3 and 64 bit

the error befor comes from -nobuffer option

thank you for the great work

spooky

spookyfirehorse, 2023-12-31:
working