jc-kynesim/rpi-ffmpeg #68: Ras Pi as RTP Server: ffmpeg error, patch available, back port possible?

saturn-hh, 2023-01-08:
I set up my Raspberry Pi as a desktop system and want to remotely access the system via RTP (freertp). Unfortunately there is a bug in ffmpeg which leads to an error ("resource temporarily unavailable") if the RTP client is a Windows system. With remmina under Linux as client, everything is fine.

The bug is solved with this patch in ffmpeg 5.2-dev:
[https://github.com/FFmpeg/FFmpeg/commit/3fc72c9fc16df3b7edc329ed9eeaaa1e65976c33](url)

I've tried to apply the patch in ffmpeg 4.4 without luck (I'm not a developper :-(  . I would like to ask if someone can back port the patch to version 4.4. 

Thank you in advance

saturn-hh, 2023-01-21:
I've tried rpi-ffmpeg-5.1 but error persist "Ressource temporary not available") . It seems that it is fixed in ffmpeg-5.2-dev.

I've got rpi-ffmpeg-5.1 with the patch compiled but the error persists due to the fact that I don't understand what is realy going on. I'm not a developer.

jc-kynesim, 2023-01-21:
The link given goes to what looks like a v4l2m2m fix (which I'm pretty sure is covered by other stuff in my branch). I'm unsure how that relates to RTP in this case?

saturn-hh, 2023-01-23:
You are right. I got the build of rpi-master working and the code of the patch is included in that build. Unfortunately the error persists.

I've tried different versions of ffmpeg and Freerdp2 but its only working with software encoding libx264 or with remmina as client. In every other combination of hardware encoding (v4l2_m2m) and a client on windows 10 I get "resource temporary not available".

Right now I try to get it working on raspberry pi os legacy with the proprietary hardware supported encoding.

saturn-hh, 2023-01-23:
How to disable encoding profiles? If I only want the profile "high level 3.2" to be included in v4l2_m2m, is there a setti g in the build process?

jc-kynesim, 2023-01-23:
Can you give me a bit (a lot would be better) more detail on your error, with logs or something? I need a rather better idea as to where the error is coming from & what might be causing it to stand any chance.

jc-kynesim, 2023-01-23:
To a large extent you get what the h/w encoder gives you - there aren't a lot of functional controls.

saturn-hh, 2023-06-16:
The bug have been on the freeRDP site and is almost solved in the code off the master branch.

On my rpi system I still can't connect with the windows rdp client but it's working with apache guacamole. So I close this issue.

Thank you again for your support.

closed 2023-06-16 by saturn-hh