jc-kynesim/rpi-ffmpeg #72: Mainlining?

Headcrabed, 2023-07-05:


jc-kynesim, 2023-07-05:
You are right that I should get round to trying to do that, but as it stands there isn't a hope in hell that it would be accepted - not because it doesn't work - it does - but because it is written as a separate thing that is just attached to ffmpeg due to having started life as a vaapi-v4l2 shim that I'd previously written so it obeys nearly none of ffmpegs normal conventions.
But yes - I should get round to reworking it into a better shape for upstreaming.

Headcrabed, 2023-07-05:
Thanks for reply. I also asked @jernejsk about this, who has another implementation of v4l2-request support. He says that your implementation has better performance (async vs lockstep) and would wait you to upstream basic HEVC stuff, then he or other LibreELEC devs would add more support for that. Maybe you could cooperate with them on this?

Headcrabed, 2023-09-26:
Hello, any progress on this?

harryqt, 2025-02-28:
It's very quiet here.

Headcrabed, 2025-03-21:
@jc-kynesim ping