jc-kynesim/rpi-ffmpeg #76: deinterlacer NV12M support

rellla, 2023-11-08:
Hi jc-kynesim,

are there any plans to support NV12M within the deinterlacer? https://github.com/jc-kynesim/rpi-ffmpeg/commit/e7a62226f26073149d35c89268f56e17c8f45d76

regards, rellla

jc-kynesim, 2023-11-08:
Not currently 'cos the Pi doesn't support that (so I'm not sure why I wrote the comment s.t. it implies I might be going to). Do you have a use case? if so I might be persuaded to write some code I can't test.

rellla, 2023-11-08:
LibreELEC is using your code for Amlogic devices
https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/multimedia/ffmpeg/package.mk#L20
wich works fine for the decoder, but iirc it's using NV12M and so we have no hardware decoder yet. This is one of the (few) missing features. But i'm not sure if all kernel bits are already there...