jc-kynesim/rpi-ffmpeg #87: Compilation failed with vulkan-headers > 1.3.277

tartanpion, 2024-03-18:
Manjaro arm with vulkan-headers 1.3.279. (vulkan_video_codec_av1std.h and vulkan_video_codec_av1std_decode.h inside).

Compilation failed with error at libavcodec/vulkan_av1.o since build: Update build.gn with AV1 headers [https://github.com/KhronosGroup/Vulkan-Headers/commit/5ac36269f50381bdd92a5e1973d8eb041771e59e](url)

CC	libavcodec/vulkan.o
CC	libavcodec/vulkan_av1.o
CC	libavcodec/vulkan_decode.o
In file included from libavcodec/vulkan_av1.c:21:
libavcodec/vulkan_decode.h:31:5: error: unknown type name 'VkVideoDecodeAV1ProfileInfoMESA'
   31 |     VkVideoDecodeAV1ProfileInfoMESA av1_profile;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:38:5: error: unknown type name 'StdVideoAV1MESATile'
   38 |     StdVideoAV1MESATile            tiles[MAX_TILES];
      |     ^~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:39:5: error: unknown type name 'StdVideoAV1MESATileList'
   39 |     StdVideoAV1MESATileList        tile_list;
      |     ^~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:43:5: error: unknown type name 'VkVideoDecodeAV1DpbSlotInfoMESA'
   43 |     VkVideoDecodeAV1DpbSlotInfoMESA    vkav1_ref;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:44:5: error: unknown type name 'StdVideoAV1MESAFrameHeader'
   44 |     StdVideoAV1MESAFrameHeader         av1_frame_header;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:45:5: error: unknown type name 'VkVideoDecodeAV1PictureInfoMESA'
   45 |     VkVideoDecodeAV1PictureInfoMESA    av1_pic_info;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:49:5: error: unknown type name 'VkVideoDecodeAV1DpbSlotInfoMESA'
   49 |     VkVideoDecodeAV1DpbSlotInfoMESA     vkav1_refs[AV1_NUM_REF_FRAMES];
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:58:29: error: unknown type name 'VkVideoDecodeAV1DpbSlotInfoMESA'; did you mean 'VkVideoDecodeAV1DpbSlotInfoKHR'?
   58 |                             VkVideoDecodeAV1DpbSlotInfoMESA *vkav1_ref, /* Goes in ^ */
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                             VkVideoDecodeAV1DpbSlotInfoKHR
libavcodec/vulkan_av1.c: In function 'vk_av1_create_params':
libavcodec/vulkan_av1.c:113:5: error: unknown type name 'StdVideoAV1MESASequenceHeader'; did you mean 'StdVideoAV1SequenceHeader'?
  113 |     StdVideoAV1MESASequenceHeader av1_sequence_header;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     StdVideoAV1SequenceHeader
In file included from libavcodec/vulkan_decode.c:21:
libavcodec/vulkan_decode.h:31:5: error: unknown type name 'VkVideoDecodeAV1ProfileInfoMESA'
   31 |     VkVideoDecodeAV1ProfileInfoMESA av1_profile;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_av1.c:114:5: error: unknown type name 'VkVideoDecodeAV1SessionParametersAddInfoMESA'; did you mean 'VkVideoDecodeH265SessionParametersAddInfoKHR'?
  114 |     VkVideoDecodeAV1SessionParametersAddInfoMESA av1_params_info;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     VkVideoDecodeH265SessionParametersAddInfoKHR
libavcodec/vulkan_av1.c:115:5: error: unknown type name 'VkVideoDecodeAV1SessionParametersCreateInfoMESA'; did you mean 'VkVideoDecodeAV1SessionParametersCreateInfoKHR'?
  115 |     VkVideoDecodeAV1SessionParametersCreateInfoMESA av1_params;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     VkVideoDecodeAV1SessionParametersCreateInfoKHR
libavcodec/vulkan_av1.c:120:28: error: 'StdVideoAV1MESASequenceHeader' undeclared (first use in this function); did you mean 'StdVideoAV1SequenceHeader'?
  120 |     av1_sequence_header = (StdVideoAV1MESASequenceHeader) {
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                            StdVideoAV1SequenceHeader
libavcodec/vulkan_av1.c:120:28: note: each undeclared identifier is reported only once for each function it appears in
libavcodec/vulkan_av1.c:120:58: error: expected ';' before '{' token
  120 |     av1_sequence_header = (StdVideoAV1MESASequenceHeader) {
      |                                                          ^~
      |                                                          ;
libavcodec/vulkan_decode.c: In function 'get_video_profile':
libavcodec/vulkan_decode.c:53:40: error: 'VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_PROFILE_INFO_MESA' undeclared (first use in this function); did you mean 'VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_PROFILE_INFO_KHR'?
   53 |         codec_id == AV_CODEC_ID_AV1  ? VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_PROFILE_INFO_MESA :
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                        VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_PROFILE_INFO_KHR
libavcodec/vulkan_decode.c:53:40: note: each undeclared identifier is reported only once for each function it appears in
libavcodec/vulkan_decode.c: At top level:
libavcodec/vulkan_decode.c:666:38: error: unknown type name 'VkVideoDecodeAV1CapabilitiesMESA'; did you mean 'VkVideoDecodeAV1CapabilitiesKHR'?
  666 |                                      VkVideoDecodeAV1CapabilitiesMESA *av1_caps,
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      VkVideoDecodeAV1CapabilitiesKHR
[... 199 lines cut ...]
libavcodec/vulkan_decode.c:914:32: error: request for member 'flags' in something not a structure or union
  914 |     dec->external_fg = av1_caps.flags & VK_VIDEO_DECODE_AV1_CAPABILITY_EXTERNAL_FILM_GRAIN_MESA;
      |                                ^
make: *** [ffbuild/common.mak:81: libavcodec/vulkan_av1.o] Error 1
make: *** Waiting for unfinished jobs....
libavcodec/vulkan_decode.c:914:41: error: 'VK_VIDEO_DECODE_AV1_CAPABILITY_EXTERNAL_FILM_GRAIN_MESA' undeclared (first use in this function)
  914 |     dec->external_fg = av1_caps.flags & VK_VIDEO_DECODE_AV1_CAPABILITY_EXTERNAL_FILM_GRAIN_MESA;
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libavcodec/vulkan_decode.c: In function 'ff_vk_decode_init':
libavcodec/vulkan_decode.c:1121:5: error: unknown type name 'VkVideoDecodeAV1SessionParametersCreateInfoMESA'; did you mean 'VkVideoDecodeAV1SessionParametersCreateInfoKHR'?
 1121 |     VkVideoDecodeAV1SessionParametersCreateInfoMESA av1_params = {
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     VkVideoDecodeAV1SessionParametersCreateInfoKHR
libavcodec/vulkan_decode.c:1122:9: error: field name not in record or union initializer
 1122 |         .sType = VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_SESSION_PARAMETERS_CREATE_INFO_MESA,
      |         ^
libavcodec/vulkan_decode.c:1122:9: note: (near initialization for 'av1_params')
libavcodec/vulkan_decode.c:1122:18: error: 'VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_SESSION_PARAMETERS_CREATE_INFO_MESA' undeclared (first use in this function); did you mean 'VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_SESSION_PARAMETERS_CREATE_INFO_KHR'?
 1122 |         .sType = VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_SESSION_PARAMETERS_CREATE_INFO_MESA,
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  VK_STRUCTURE_TYPE_VIDEO_DECODE_AV1_SESSION_PARAMETERS_CREATE_INFO_KHR
cc1: some warnings being treated as errors
make: *** [ffbuild/common.mak:81: libavcodec/vulkan_decode.o] Error 1
==> ERROR: A failure occurred in build().
    Aborting...

jc-kynesim, 2024-03-19:
N.B. The 5.1.4 in the above comment assumes that is the branch of mine you were building - if you are building something else please specify (and check against the corresponding mainline branch).

tartanpion, 2024-03-19:
More background : rpi5. The problem has occurred when updating vulkan-headers 1.3.276 > 1.3.279 with vulkan av1 header included.
With source='git+https://github.com/jc-kynesim/rpi-ffmpeg.git#branch=dev/6.1/rpi_import_1' or 6.1.1/rpi_import and --enable-vulkan.

> Enabled hwaccels:
av1_vaapi               h264_vaapi              hevc_vaapi              mpeg2_vaapi             vp8_vaapi
av1_vulkan              h264_vulkan             hevc_vulkan             mpeg4_vaapi             vp9_vaapi
h263_vaapi              hevc_v4l2request        mjpeg_vaapi             vc1_vaapi               wmv3_vaapi

If i configure with --disable-vulkan as a workaround no problem.

I have tried with main/5.1.4 (--enable-vulkan enabled) and no problem because it doesn't do anything with vulkan :

> CC	libavcodec/vp9data.o
CC	libavcodec/vp9dsp.o
CC	libavcodec/vp9dsp_10bpp.o
CC	libavcodec/vp9dsp_12bpp.o
CC	libavcodec/vp9dsp_8bpp.o
CC	libavcodec/vp9lpf.o
CC	libavcodec/vp9mvs.o
CC	libavcodec/vp9prob.o
CC	libavcodec/vp9recon.o
CC	libavcodec/vqavideo.o
CC	libavcodec/wavpack.o
CC	libavcodec/wavpackdata.o

> Enabled hwaccels:
av1_vaapi               hevc_v4l2request        mpeg2_vaapi             vp8_vaapi
h263_vaapi              hevc_vaapi              mpeg4_vaapi             vp9_vaapi
h264_vaapi              mjpeg_vaapi             vc1_vaapi               wmv3_vaapi

jc-kynesim, 2024-03-20:
OK - If you have the issue with 6.1.1/rpi_import then I'm going to bet the problem exists upstream too. I'm going with "not my problem" till proven otherwise. If you want to submit a patch that fixes the issue and doesn't break compiles with earlier headers then feel free and I'll include it in my tree.

tartanpion, 2024-03-25:
[https://github.com/FFmpeg/FFmpeg/commit/fef22c87ada4517441701e6e61e062c9f4399c8e](https://github.com/FFmpeg/FFmpeg/commit/fef22c87ada4517441701e6e61e062c9f4399c8e)

closed 2024-03-25 by tartanpion