jc-kynesim/rpi-ffmpeg #89: raspberry pi 1

HTMofDreams, 2024-04-17:
Witch version I have to use, for an old raspberry pi 1?

jc-kynesim, 2024-04-17:
The version used for PiOS test/5.1.4/main is probably your best bet.

jc-kynesim, 2024-04-17:
It depends what you are trying to do. The options set up by pi-util/conf_native.sh are what I use, but those assume (I think) at least a Pi3 (maybe 2). You'll have to tweak the cpu/arch options for a Pi1, and I can't offhand remember the correct answers.
Is there some reason not to use the PiOS binary?

HTMofDreams, 2024-04-17:
Its only a hobby. In the moment I use retropie/debian buster with kodi, to watch a little bit iptv. Of course, I  tested raspbian/bookworm, too. But kodi works much more faster with retropie/debian buster. I have backported kodi 19.1 to retropie, because there is no kodi 19.1 in the repositories. Now, kodi 19.1 works, but its very slow to play any videos. The debian maintainer of kodi tells me, I have to use a pachted ffmpeg version for the raspberry pi 1.

jc-kynesim, 2024-04-17:
On a Pi1 you need h/w decode all the way to display without going anywhere near the ARM to get useful decode. I'd have expected raspbian Kodi to work OK. This is one of those things that has not got easier with every upgrade of PiOS.

lurch, 2024-04-17:
> On a Pi1 you need h/w decode all the way to display without going anywhere near the ARM to get useful decode.

...which I believe might require https://www.raspberrypi.com/documentation/computers/config_txt.html#licence-key-and-codec-options depending on the video-format that you're trying to play?

popcornmix, 2024-04-17:
Latest kodi is not usable on Pi1. It doesn't have the RAM needed.
And omxplayer support has been removed which was the one option that worked on the weedy pi 1.

HTMofDreams, 2024-04-17:
> Latest kodi is not usable on Pi1. It doesn't have the RAM needed.
> And omxplayer support has been removed which was the one option that worked on the weedy pi 1.

I had thought of that too.

> ...which I believe might require https://www.raspberrypi.com/documentation/computers/config_txt.html#licence-key-and-codec-options depending on the video-format that you're trying to play?

My videos all in h264. I always thought, that h264 is build in, in the raspberry pi 1.

jc-kynesim, 2024-08-13:
Sorry - don't think this is really an ffmpeg issue

closed 2024-08-13 by jc-kynesim