jc-kynesim/rpi-ffmpeg #96: backwards playback/ slow motion - internal decoder buffers

rellla, 2024-12-18:
Hi,

i'm using test/6.0.1/main atm and want to solve the following issue in a player/output plugin. Platform is a RPI4.

Situation:
- Playing videos in forward direction works just fine
- Backwards slow motion or fast rewind is implemented in the player by sending AVPackets (only I-Frames) with a PTS in decreasing order. 

In normal forward playback it's no problem to send AVPackets until a decoded frame is ready, because the decoder sorts (and outputs) the decoded frames with increasing PTS, because we play forward. Also it's no problem, if we have more than 1 AVPackets internally buffered.
In backward playback (slow or fast rewind), sending more AVPackets is a problem, because the decoder doesn't perform the intended sorting. For mpeg2 this seems to be ok, but for h264 interlaced material, 11 AVPackets have to be sent to get the first frame, which isn't the AVPacket which was sent first (with the greatest PTS). The next one, we get with avcodec_receive_frame() is the one, which has the next PTS in forward direction...

I had similiar problems with rockchip (v4l2request), too. There it is solved be doing the following logic:
send AVPacket
try receive AVFrame
if we received a frame, flush the decoder, send the next AVPacket

My implementation is https://github.com/rellla/vdr-plugin-softhddevice-drm-gles/blob/drm-atomic-gles/softhddev.c#L1149 atm.

This works for rockchip (and mpeg2) just fine - though i need to restart the decoder after a normal playback, because the expected AVPackets for a AVFrame could be > 1, if we had some B-Frames before. I'm getting a frame as soon as possible. With rpi4 mpeg2 there is a problem with this implementation, because the decoder wants 2 AVPacket, sends 1 AVFrame, I do the flush and the second one gets lost... Though i have to investigate the mpeg2 thing when software decoding on rockchip once more, i want to get h264 solved.

So my question is:
Is there any flag or hardcoded value in your code, which tells the h264_v4l2m2m decoder to wait for .... AVPackets before sending the first decoded frame? Could the decoder be forced to return the AVFrame as soon as it got all the needed AVPackets (we have just I-Frames of course, so 2 interlaced AVPackets should be enough...)?

I hope, i explained my problem in a way you can understand it ;) Thanks for your help.

Regards
Andreas

jc-kynesim, 2024-12-18:
Yeah I understand the issue. Honestly I'd use the s/w decoder which will probably be a lot better behaved. To force out the packet that you want you should send an EndOfStream (i.e. a null packet), however I'm not honestly sure if that will work (it should) but I think you may need to restart the decoder afterwards which is going to take way more time than just using the s/w decode.
The other thing you could try is rewriting the SPS to have no reordering and a single entry DPB.

rellla, 2025-01-16:
Hi, i did try to understand ffmpeg in the meanwhile. I did some changes in the code just to see the effects, but i couldn't get the decoder to return a frame, until the buffer was filled at least with 10 packets (for H264). As long as i feed it with less than 10 packets i get the EAGAIN with avcodec_receive_frame. I couldn't find the code, which is responsible for this. Can you give me a hint, where i could have a look?

Sending the EOS works kind of. But only if the decoder ist restarted. According to https://www.ffmpeg.org/doxygen/7.0/group__lavc__encdec.html a avcodec_flush_buffers() should be enough. Can you tell me, why this does not work here? I still get EOF from avcodec_receive_frame() after avcodec_flush_buffers(). Would be glad, if you can point me to some code ...

If rewriting the SPS would be worth it, where is the reordering or the DPB handled in the code? With rewriting SPS you mean setting sps->num_reorder_frames to 1?

Thanks for your help again.

rellla, 2025-01-17:
https://www.linuxtv.org/downloads/v4l-dvb-apis/userspace-api/v4l/dev-decoder.html
After reading this one, i come to the conclusion, that best (and maybe only reasonable) way would be to force draining by sending a null packet. 
It's not guaranteed by design, that a frame should be available as soon as possible - if i understand correctly.
So i'd like to solve the avcodec_flush_buffers issue but can use close/open the decoder as a workaround for now.

rellla, 2025-01-21:
It's me again ;)
I workaraound my issue now by closing and opening the codec after sending an EOS. avcodec_flush_buffers doesn't work and avcodec_receive_frame still returns AVERROR_EOF.

Can you confirm, that avcodec_flush_buffers() isn't working as expected? If so, would it be possible to fix it in the code? I guess, https://github.com/jc-kynesim/rpi-ffmpeg/blob/test/7.1/main/libavcodec/v4l2_m2m_dec.c#L1285 would be the place where i have to look?

jc-kynesim, 2025-01-21:
Do you have a small example of the stream you are trying to play that I can look at so I can see what latency I'd expect?

rellla, 2025-01-21:
I can provide a stream, but i'm not sure if that is helpful. It's a simple DVB-S stream as MPEG TS container with h264 or mpeg2 DVB-S data in it.
https://github.com/rellla/vdr-plugin-softhddevice-drm-gles/tree/WIP/rpiV02 is my latest code. This plugin gets the raw stream data from a player. In case of backward seek only I-Frames are sent for example.
The latest commit message tries to explain my problem.
https://github.com/rellla/vdr-plugin-softhddevice-drm-gles/blob/WIP/rpiV02/softhddev.c#L1097 is the decoding thread loop and CodecVideoFlushBuffers https://github.com/rellla/vdr-plugin-softhddevice-drm-gles/blob/WIP/rpiV02/codec.c#L431 should just be the avcodec_flush_buffers in my understanding - not a close/reopen.

The code works fine on a pi now, but i will retest it again with avcode_flush_buffers in CodecVideoFlushBuffers, just to confirm it not working. Rockchip with v4l2request did fine with avcodec_flush_buffers iirc.

I'm on test/7.1/main btw...

rellla, 2025-01-23:
I changed the code to use avcodec_flush_buffers after sending EOS and this is the log:
https://paste.debian.net/1346411/
Decoding workflow is https://github.com/rellla/vdr-plugin-softhddevice-drm-gles/blob/WIP/rpiV04/softhddev.c#L1098

Please don't bother about some additional ffmpeg logs, i added them in the code.
The workflow is as follows:

- initiate backward trickspeed
- avcodec_send_packet(ctx, packet)
- if (success) avcodec_send_packet(ctx, NULL)
- avcodec_receive_frame(ctx, &frame)
- send it to the render ringbuffer
- try again avcodec_receive_frame(ctx, &frame)
- if (AVERROR_EOF) avcodec_flush_buffers(ctx)
- try next avcodec_send_packet(ctx, packet)

After avcodec_flush_buffers i can send a new packet and force the eos again, but avcodec_receive_frame doesn't receive the frame but still reports EOF.

Is there an error in my logic, or is sth wrong with resetting EOF?

Thanks for your help again.

rellla, 2025-01-23:
I switched back to avcodec_flush_buffers and this is the log:
https://paste.debian.net/plain/1346418

Based on the following decoding logic:
https://github.com/rellla/vdr-plugin-softhddevice-drm-gles/blob/WIP/rpiV04/softhddev.c#L1098

The workflow is as follows:

- initiate backward trickspeed (player sends I-Frames in descending PTS order)
- avcodec_send_packet(ctx, packet)
- if (success) then avcodec_send_packet(ctx, NULL)
- avcodec_receive_frame(ctx, &frame)
- if (success) then send frame to render ringbuffer
- try avcodec_receive_frame again
- if (AVERROR_EOF) then avcodec_flush_buffers(ctx)
- continue with sending the next packet

The first avcodec_receive_frame returns a frame, the second one returns EOF (as expected).
After the flush i can send an AVPacket again, but avcodec_receive_frame still returns AVERROR_EOF

Is there some issue in my decoding logic, or is this an issue in avcodec_flush_buffers? 
Thanks for your help again.

rellla, 2025-02-20:
Hi,
i think i found it.
It works, when timeout isn't set to "0" here. https://github.com/jc-kynesim/rpi-ffmpeg/blob/test/7.1/main/libavcodec/v4l2_m2m_dec.c#L657

Either the "0" has to be higher, or prefer_dq needs to be set in my error case. I guess prefer_dq has to be set > 0 if EOS is sent. It isn't here. 

Would be cool, if you can have a look at that :)

jc-kynesim, 2025-02-20:
If we've received an EOS (empty) packet then src_rv should be NQ_DRAINING
which should give a decent length timeout irrespective of prefer_dq in that
statement.

rellla, 2025-02-20:
https://pastebin.com/zkmG7Mst

I will write some thoughts later, but this is my log. Lines 463 and above should be the interesting ones.

jc-kynesim, 2025-02-20:
OK - I'm not entirely sure what is going on here but we seem to be getting
EAGAINs after the EOF - I didn't think that could happen until the status
was cleared.
Try copying the block:

    if (s->draining) {
        if (s->buf_pkt.size) {
            av_log(avctx, AV_LOG_WARNING, "Unexpected input whilst draining\n");
            av_packet_unref(&s->buf_pkt);
        }
        return NQ_DRAINING;
    }

which is outside the "if(!s->buf_pkt.size && !do_not_get)" conditional in
try_enqueue_src to just before the test "if (ret == AVERROR(EAGAIN))"
inside the condition to force EOS once we've seen it once.

rellla, 2025-02-20:
I will try that and report.

rellla, 2025-02-20:
Ok, so that seems to work.
Here is the new log with the same workflow and debug code: https://pastebin.com/uZApgE9x
- starting at 341 and upwards. Now we get the src_rv 4! Not tested on RPI btw, but on amlogic, but that should be equal.
Thanks for your help.

jc-kynesim, 2025-02-21:
Hi

>rellla left a comment (jc-kynesim/rpi-ffmpeg#96)
>
>Ok, so that seems to work.
>Here is the new log with the same workflow and debug code: https://pastebin.com/uZApgE9x
>- starting at 341 and upwards. Now we get the src_rv 4! Not tested on RPI btw, but on amlogic, but that should be equal.

I was suspecting not Pi - Pi returns EOS events when requested at EOS -
what you have doesn't (which is out of spec and isn't helping here).

>Thanks for your help.

I'll add something to the effect of that patch to the tree shortly

Regards

rellla, 2025-02-21:
I can try that on RPI and iirc, it was the same. Stay tuned ...

rellla, 2025-02-21:
https://pastebin.com/6PkskwZt (line 2212 and upwards) is with the patch
https://pastebin.com/8NuGzjQu (line 1586 and up) is without the patch

Both seem to have the same problem. The first frame is returned. 
Then i try a avcodec_receive_frame again, which returns EOF, which is ok. This EOF triggers avcodec_flush_buffers.

The next avcodec_send_packet works, and the very next is triggering the drain by sending a NULL packet, which ends in a EOS request. This is also correct, i think. The problem comes now, because the frame could not be returned. It's around the Buffer empty...

This problem is probably a different one to amlogic, but i don't find the problem in my ffmpeg-api workflow at userspace side.

rellla, 2025-02-21:
one nitpick regarding the logs:
`Buffer empty - reQ is_mp 1 plane.bytesused 0 bytesused 0 flag_last 1` flag_last is 0 when entering the if. i placed the log line to the end.

rellla, 2025-02-27:
Hi @jc-kynesim ,
i did some more testing and now have a log which includes the kernel debugs:
https://gist.github.com/rellla/0ce266e0619da5a03f0ffffcc5c61b89#file-gistfile1-txt-L6128

Though userspace and kernel timestamps may not match exactly we run into some strange behaviour beginning with line 6128.

Line 6068 seems to be the eos-buffer which has a timestamp +20000 to the last. Not sure, if this is right.

May the problem be a kernelside one?

rellla, 2026-02-17:
Afaics, the necessary flush in the draining is done with off/on, but it seems there is only done a STREAMOFF on the output side:
https://github.com/jc-kynesim/rpi-ffmpeg/blob/test/8.0.1/main/libavcodec/v4l2_m2m_dec.c#L1357

Is it necessary to keep some order in off/on? Is the missing STREAMON related at all? It's done lazily later iirc.
I can't interpret the log: https://gist.github.com/rellla/0ce266e0619da5a03f0ffffcc5c61b89#file-gistfile1-txt-L6128 
What happens kernelside here between the STREAMOFF-STREAMON on the capture side?
Is it necessary to have the output side ON here?

rellla, 2026-02-17:
> If you think I'm violating what it says then please tell me and I'll fix it.

Hi, if this is related to my answer above, i'm not sure if or what is wrong either. I just came across it and noticed, that avcodec_flush_buffers is not doing what it should according to ffmpeg docs :)

jc-kynesim, 2026-02-17:
I think I'm doing the right thing and the kernel/firmware is getting confused somehow - it wouldn't be the first time.

As an aside, when talking about a V4L2 OUTPUT please (for my benefit) capitalize it so I am sure we are talking about the same thing, the V4L2 input=OUTPUT, output=CAPTURE thing is just so confusing. I know ffmpeg uses _output and I can't help that (in code I've written I try to use _src and _dst to avoid confusion). 

But thank you for the report - I'm just not sure how to fix the problem.

rellla, 2026-10-19:
@jc-kynesim Feeding Claude with the infos, it generated this patch 
https://gist.github.com/rellla/a592fb77dcd72f9d1047db2cfdc0a90b
and it really seems to solve this problem. Maybe you can have a look at it ;)

jc-kynesim, 2026-10-19:
Claude may have a point. I'm not sure I love his fix, but I'll definitely consider it or something like it.

P.S. the pastebin links you posted back in Feb 25 seems to have some sort of Mal/Adware stuff that wants me to enable notifications attached to it now :-( so I can't/won't go back and check if what claude says matches the debug.

rellla, 2026-10-19:
Yeah, sth is wrong with pastebin, i got redirected to a malware site and so used gist instead. I would be happy, if you can consider thinking about what Claude says. The patch works here - i don't need to do the codec close/reopen workaround instead of the flush anymore. Sure, you may use your own patch - Claude is highly communicative ....

Btw: I just sent him the single rpi-ffmpeg-patch from LE and my statement in https://github.com/jc-kynesim/rpi-ffmpeg/issues/96#issuecomment-3896392270

rellla, 2026-10-19:
Wait, my bad. I tested the wrong file.... The patch does NOT fix it. I got another hint for an issue at kernel side and will test that one. Sorry for the noise!

rellla, 2026-10-21:
Hi @jc-kynesim and @6by9 ,

I tracked this down with the help of Claude, after doing some tests with different patches, this is his explanation:

------------------------------
It's not ffmpeg, it's the bcm2835-codec kernel driver plus a firmware limitation. The original ffmpeg flush code (STREAMOFF/STREAMON on both queues after a drain) is fine.

There are three separate issues:

1. **stop_streaming doesn't wait for the VPU buffers (driver bug).**
   `ip_buffer_cb()`/`op_buffer_cb()` only signal `frame_cmplt` while `buffers_with_vpu` is still non-zero. The MMAL core decrements the counter before calling the callback, so the waiter is woken by the first returned buffer and never by the last one. That's where the `driver bug: stop_streaming operation is leaving buffer N in active state` warnings in my earlier trace come from. The fix is to invert the check (`!atomic_read(...)`).

2. **The EOS can overtake the last OUTPUT buffer (driver bug).**
   `V4L2_DEC_CMD_STOP` submits the EOS buffer to MMAL directly, while OUTPUT buffers are only passed on from `device_run()`. If the drain arrives before `device_run()` has picked up the queued packet, the VPU gets the EOS first. The frame is then never decoded, no EOS comes back on CAPTURE, and ffmpeg ends up in "Timeout in drain". The fix is to defer the EOS until the m2m OUTPUT queue is empty.

3. **After a drain the firmware decoder doesn't decode anymore (firmware).**
   Even with 1 and 2 fixed, every drain after the first one only returns the empty EOS buffer, no frame. Restarting both queues (which disables and re-enables the component) doesn't help, and neither does an MMAL port flush (`MMAL_MSG_PORT_ACTION_TYPE_FLUSH` on both ports is accepted but changes nothing). Only a freshly created component works again, which is why closing and reopening the decoder was the only workaround. As a workaround in the driver, I release `ril.video_decode` once both queues are stopped after a drain and recreate it on the next start_streaming.
------------------------------
Patches:
For 1: [0001-staging-bcm2835-codec-Fix-inverted-check-when-waitin.patch](https://gist.githubusercontent.com/rellla/3bf4bd76ae0cc3e28eefc4699b064952/raw/0c34a95a20adefd6a56f3545f391d9efedaa9c6b/gistfile1.txt)
For 2: [0003-staging-bcm2835-codec-Don-t-let-the-decoder-EOS-over.patch](https://gist.githubusercontent.com/rellla/7b2274d25abb66bdfbfd306de0052b4d/raw/95c01d5d35fe5972f0e812a18a0225ae5fff13c9/gistfile1.txt)
For 3: [0002-staging-bcm2835-codec-Recreate-decoder-component-aft.patch](https://gist.githubusercontent.com/rellla/34ad019460190024c2c956ea8966e054/raw/9708641dcf12f85741fa74de61d436e66b52a5a5/gistfile1.txt)

With all three patches (against rpi-6.18.y), trick play with `send_packet(avpkt)` + `send_packet(NULL)` + `receiving AVERROR_EOF on both sides` + `avcodec_flush_buffers()` works reliably on the RPi4 without reopening the decoder.

What do you think of it?

jc-kynesim, 2026-10-21:
Given that I could not see anything in the debug that suggested that I was doing something wrong I'm happy for it to be a driver issue! @6by9 what do you think?

6by9, 2026-10-21:
The first two looks reasonable.

The third is a sledgehammer. Is there a simple test case I can run to trigger this behaviour? I'm guessing it's a resolution change mid-stream that is required. I can then see what the firmware is up to.

rellla, 2026-10-21:
The third is probably a workaround for wrong firmware behaviour. Though not a primitive setup, my test case is VDR with https://github.com/rellla/vdr-plugin-drmdevice as an output plugin handling decoding and rendering.
https://github.com/rellla/vdr-plugin-drmdevice/blob/main/videostream.cpp#L433 is the starting point.
Without the above patches (all 3) i need this workaround: https://github.com/rellla/vdr-plugin-drmdevice/blob/main/videostream.cpp#L186
All other platforms (rockchip, allwinner, amlogic) show right behaviour. 
On backward trickspeed, VDR as the "receiver" sends I-Frames in descending PTS order. I need to force the drain, because otherwise the decoder does some reordering and other unexpected things.
Imho the plugins uses correct FFmpeg API, rpi-mpeg seems to right, but kernel (esp. firmware) does miss some reset after the first EOS. The patch just seems to workaround this.

rellla, 2026-10-21:
Maybe i can advise Claude to design a small standalone-test with the plugin's behaviour if you want?

rellla, 2026-10-21:
@6by9 Claude wrote a standalone tool for me, which simulates what happens in my player:
https://github.com/rellla/vdr-trickrev

All you need is a h264 video file with dedicated I-Frames. I tested with a h264 DVB-S2 TS file. If you want, i can upload a snippet somewhere. 

Claude created a fine readme for it. Building it against rpi-ffmpeg on your system and using your kernel should just work fine. I tested building on a RPI OS and on VDRSternELEC (which is a LibreELEC-hook-up). I did not yet test it against the kernel without patches, but you can do that against the current 6.18 for example with:

`./vdr-trickrev --speed 3 --flush flush --count 20 00001.ts`
`./vdr-trickrev --speed 3 --flush reopen --count 20 00001.ts`

If all works as I expect, the first shouldn't work as expected, the second should.
And as the next step, test this with the above patches. Both tests should work.