jc-kynesim/rpi-ffmpeg #97: h264_v4l2m2m / bcm2835_v4l2 hardware accelerated decoding on kernel 6.6.63

t3therdev, 2024-12-18:
Hi All,

We are trying to resolve an issue related to migrating the pi4/cm4 hardware from kernel 5.15 / ffmpeg 5.1.x where hardware decoding worked perfectly.  To kernel 6.6.63/FFMPEG 7 (tested on mainline and your branch /test/7.1/main and /test/6.01/main). On the new kernel/ffmpeg hardware decoding specifically does not work and just hangs. If we remove the hardware decoding, encoding works perfectly. 

For example, the following works on the older setup but does not on the newer kernel.  removing the decode portion makes it work on the newer versions.

`ffmpeg -f video4linux2 -c:v h264_v4l2m2m -input_format h264 -video_size 1280x720 -framerate 30 -i /dev/video0 -c:v h264_v4l2m2m -an -pix_fmt yuv420p -b:v 2048k -r 30 /test.mp4`

(with -c:v h264_v4l2m2m removed)

`ffmpeg -f video4linux2 -input_format h264 -video_size 1280x720 -framerate 30 -i /dev/video0 -c:v h264_v4l2m2m -an -pix_fmt yuv420p -b:v 2048k -r 30 /test.mp4`

on the boot/config/.txt we have gpu_mem=128 and have tried other settings as well with no success. falling back to the same exact configuration on the older kernel works perfectly. 

Running ffmpeg in debug mode:
`Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-f' ... matched as option 'f' (force container format (auto-detected otherwise)) with argument 'video4linux2'.
Reading option '-c:v' ... matched as option 'c' (select encoder/decoder ('copy' to copy stream without reencoding)) with argument 'h264_v4l2m2m'.
Reading option '-input_format' ... matched as AVOption 'input_format' with argument 'h264'.
Reading option '-video_size' ... matched as AVOption 'video_size' with argument '1280x720'.
Reading option '-framerate' ... matched as AVOption 'framerate' with argument '30'.
Reading option '-i' ... matched as input url with argument '/dev/video0'.
Reading option '-fflags' ... matched as AVOption 'fflags' with argument 'nobuffer'.
Reading option '-c:v' ... matched as option 'c' (select encoder/decoder ('copy' to copy stream without reencoding)) with argument 'h264_v4l2m2m'.
Reading option '-an' ... matched as option 'an' (disable audio) with argument '1'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'yuv420p'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '2048k'.
Reading option '-r' ... matched as option 'r' (override input framerate/convert to given output framerate (Hz value, fraction or abbreviation)) with argument '30'.
Reading option '-f' ... matched as option 'f' (force container format (auto-detected otherwise)) with argument 'rtsp'.
Reading option 'rtsp://stream:stream@localhost:554/device1' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url /dev/video0.
Applying option f (force container format (auto-detected otherwise)) with argument video4linux2.
Applying option c:v (select encoder/decoder ('copy' to copy stream without reencoding)) with argument h264_v4l2m2m.
Successfully parsed a group of options.
Opening an input file: /dev/video0.
[video4linux2,v4l2 @ 0x7fb5c3c0d0] fd:3 capabilities:84a00001
[video4linux2,v4l2 @ 0x7fb5c3c0d0] Current input_channel: 0, input_name: Camera 1, input_std: 0
[video4linux2,v4l2 @ 0x7fb5c3c0d0] Setting time per frame to 1/30
[h264 @ 0x450320] non-existing PPS 0 referenced
[extract_extradata @ 0x7fb3dd9db0] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] non-existing PPS 0 referenced
[h264 @ 0x450320] decode_slice_header error
[h264 @ 0x450320] no frame!
[h264 @ 0x450320] non-existing PPS 0 referenced
[extract_extradata @ 0x7fb3dd9db0] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x450320] non-existing PPS 0 referenced
[h264 @ 0x450320] decode_slice_header error
[h264 @ 0x450320] no frame!
[h264 @ 0x450320] Decoding VUI
[extract_extradata @ 0x7fb3dd9db0] nal_unit_type: 7(SPS), nal_ref_idc: 3
[extract_extradata @ 0x7fb3dd9db0] nal_unit_type: 8(PPS), nal_ref_idc: 3
[extract_extradata @ 0x7fb3dd9db0] nal_unit_type: 5(IDR), nal_ref_idc: 3
[h264 @ 0x450320] nal_unit_type: 7(SPS), nal_ref_idc: 3
[h264 @ 0x450320] nal_unit_type: 8(PPS), nal_ref_idc: 3
[h264 @ 0x450320] nal_unit_type: 5(IDR), nal_ref_idc: 3
[h264 @ 0x450320] Decoding VUI
[h264 @ 0x450320] Format yuvj420p chosen by get_format().
[h264 @ 0x450320] Reinit context to 1280x720, pix_fmt: yuvj420p
[h264 @ 0x450320] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
    Last message repeated 5 times
[video4linux2,v4l2 @ 0x7fb5c3c0d0] All info found
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 7086.841769, bitrate: N/A
[... 40 lines cut ...]
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-isp' on card 'bcm2835-isp' in splane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] v4l2 output format not supported
[h264_v4l2m2m @ 0x7fb3eb4c00] probing device /dev/video12
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-codec' on card 'bcm2835-codec-isp' in mplane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] v4l2 output format not supported
[h264_v4l2m2m @ 0x7fb3eb4c00] probing device /dev/video20
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-isp' on card 'bcm2835-isp' in splane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] v4l2 output format not supported
[h264_v4l2m2m @ 0x7fb3eb4c00] probing device /dev/video10
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] Using device /dev/video10
[h264_v4l2m2m @ 0x7fb3eb4c00] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x7fb3eb4c00] requesting formats: output=H264/none capture=YU12/yuvj420p
[h264_v4l2m2m @ 0x7fb3eb4c00] output: H264 16 buffers initialized: 1280x0720, sizeimage 00691328, bytesperline 00000000
Successfully opened the file.
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_v4l2m2m) -> h264 (h264_v4l2m2m))
[vost#0:0/h264_v4l2m2m @ 0x7fb38c6820] Starting thread...
[vf#0:0 @ 0x7fb5af5650] Starting thread...
[vist#0:0/h264 @ 0x7fb8991340] [dec:h264_v4l2m2m @ 0x7fb3eb42a0] Starting thread...
[in#0/video4linux2,v4l2 @ 0x7fb5cbf420] Starting thread...
Press [q] to stop, [?] for help
[h264_v4l2m2m @ 0x7fb3eb4c00] capture: YU12 20 buffers initialized: 1280x0720, sizeimage 01382400, bytesperline 00001280
[h264 @ 0x7fb38c6080] Decoding VUI
`

jc-kynesim, 2024-12-19:
What happens with ffmpeg 5.1.6 (current PiOS version) on the newer kernel?

t3therdev, 2024-12-19:
hi @jc-kynesim mainline 5.1.4 and 5.1.6 exhibit the same issues. 

we will give this a try as well:
https://github.com/jc-kynesim/rpi-ffmpeg/tree/test/5.1.6/main

it does seem likes its a kernel level issue. are you able to do it on a pi4 on any of your builds?

t3therdev, 2024-12-20:
As a quick follow up we tested on rpi 3 hardware and it has the same issue. So assuming kernel 6.6 related.

jc-kynesim, 2024-12-20:
Seems odd - I'll have a look when I have a moment.

6by9, 2024-12-20:
I've just posted on the duplicate - https://github.com/raspberrypi/linux/issues/6554#issuecomment-2557037733

Decoding of Big Buck Bunny works fine
ffmpeg -c:v h264_v4l2m2m -i bbb_sunflower_1080p_60fps_normal.mp4 -an -pix_fmt yuv420p foo.yuv

Transcode of the 1080p60 version fails as ffmpeg doesn't set the level, and the default of 4.0 isn't sufficient for 1080p60.
Use 720p60 clip and it's fine.
ffmpeg -c:v h264_v4l2m2m -i Big_Buck_Bunny_720_10s_5MB.mp4 -c:v h264_v4l2m2m -an -pix_fmt yuv420p -b:v 2048k test.mp4
(file from https://test-videos.co.uk/vids/bigbuckbunny/mp4/h264/720/Big_Buck_Bunny_720_10s_5MB.mp4)

This is on a fresh install of Raspberry Pi OS Bookworm (19/11/2024), kernel 6.6.51+rpt-rpi-v8, ffmpeg version 5.1.6-0+deb12u1+rpt1, on a Pi4.

rpi-update to get 6.6.66, and still works fine.

I suspect we need a sample stream from your webcam.

t3therdev, 2024-12-21:
@6by9 thanks for your feedback. We were able to confirm that hardware decode on the suggested file does work. Which is good news. 

I have attached 2 sample raw captures from our usb camera that do not work. Hopefully this can provide some insight as to the issue and why it worked on kernel 5.15 but not 6.6.

[output_samples.zip](https://github.com/user-attachments/files/18217011/output_samples.zip)

jc-kynesim, 2024-12-24:
Still odd. Your sample file plays fine for me just replacing the video capture bit of the command line with the capture file. Neither of the USB cameras I have here like the setup you are using (they don't want to output h264) so a direct comparison can't be done right now. This'll have to wait to the new year when I can scrounge a camera that looks a bit more like yours.

t3therdev, 2024-12-28:
@jc-kynesim @6by9 thanks for your feedback. we have placed a stream on a temporary vps for testing. this rtsp stream with h264 source exhibits the same issue, so a usb camera is not needed.

Non working
> ffmpeg -c:v h264_v4l2m2m -rtsp_transport tcp -i rtsp://199.127.61.234:9094/cam1 -c:v h264_v4l2m2m -pix_fmt yuv420p -b:v 2048k -an test.mp4

Removing the hw decode, and working
> ffmpeg -c:v h264_v4l2m2m -rtsp_transport tcp -i rtsp://199.127.61.234:9094/cam1 -c:v h264_v4l2m2m -pix_fmt yuv420p -b:v 2048k -an test.mp4

6by9, 2025-01-03:
Vanilla Raspberry Pi OS Bookworm install, although I have updated the kernel to 6.6.69 due to current development work. No significant changes to config.txt or cmdline.txt.

Hardware decode of that stream using
`ffmpeg -c:v h264_v4l2m2m -rtsp_transport tcp -i rtsp://199.127.61.234:9094/cam1 -an -pix_fmt yuv420p foo.yuv`
works fine.

`ffmpeg -c:v h264_v4l2m2m -rtsp_transport tcp -i rtsp://199.127.61.234:9094/cam1 -c:v h264_v4l2m2m -pix_fmt yuv420p -b:v 2048k -an test.mp4`
also works fine.

```
pi@pi:~ $ uname -a
Linux pi 6.6.69-v8+ #2 SMP PREEMPT Fri Jan  3 12:01:28 GMT 2025 aarch64 GNU/Linux
pi@pi:~ $ vcgencmd version
Aug 30 2024 19:17:39 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)
pi@pi:~ $ ffmpeg -version
ffmpeg version 5.1.6-0+deb12u1+rpt1 Copyright (c) 2000-2024 the FFmpeg developers
built with gcc 12 (Debian 12.2.0-14)
```

Looking at your log in more detail, you have yuvj420p as the image format in places. AIUI That is full range yuv420 as used by JPEG, whereas H264 normally uses limited range. I wonder if ffmpeg is trying to convert all the images between the two ranges, and is therefore overloading a CPU, or just failing.

t3therdev, 2025-01-03:
Hi @6by9 , so using your exact commands we still are not able to use the hardware decoder. removing the hardware decode reference does work. i think we were only using the yuv420p to feed the hardware encoder (which is working). 

confirmed again the previous kernel that hardware decoding does work. there has to be something with this kernel / openwrt combination that we are missing. 

> Linux 6.6.63 #0 SMP Tue Dec 10 00:22:05 2024 aarch64 GNU/Linux

> Sep 13 2024 15:58:42
Copyright (c) 2012 Broadcom
version ddfba3e3c234500025b545512b4b214f28e453e9 (clean) (release) (start)

> ffmpeg version 5.1.6 Copyright (c) 2000-2024 the FFmpeg developers
built with gcc 13.3.0 (OpenWrt GCC 13.3.0 r28178-28de3ffefa)

6by9, 2025-01-04:
> I was able to confirm on bookworm standard release its working with the camera using:
> 
> > Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux
> 
> So the issue still is persisting on Openwrt / Rpi4 with even the latest kernel on 24.10 RC4.

Sorry, but if it is only an issue on some other distribution (first time you've actually stated that), then it's not something that we're likely to investigate.