jc-kynesim/rpi-ffmpeg #98: test/7.1.1/main tries to use vulkan when disabled

juanitotc, 2025-05-09:
I built branch test/7.1.1/main to try hevc hardware decoding on an RPi5 using kernel 6.12.25 with the rpi-hevc-dec driver.

The --disable-vulkan configure switch was used, but despite this, ffmpeg tries to use vulkan - am I missing something?

This works:
```
SDL_VIDEODRIVER=kmsdrm ffplay -i /mnt/mmcblk0p2/media/video/*hevc*
```

This does not work:
```
ffmpeg -hwaccels
drm 

SDL_VIDEODRIVER=kmsdrm ffplay -hwaccel drm -i /mnt/mmcblk0p2/media/video/*hevc*
...
Enable vulkan renderer to support hwaccel drm
Doesn't support vulan renderer, fallback to SDL renderer
...
Vulkan renderer is not available
Failed to open file /mnt/mmcblk0p2/media/video/jellyfish-3-mbps-hd-hevc.mkv or configure filtergraph
```

jc-kynesim, 2025-05-10:
The issue isn't with decode, the issue is with display. The problem is that ffplay doesn't have an appropriate renderer to deal with the drm_prime buffers created by the decoder - and if you want anything like smooth playback on a Pi you want h/w support at the render stage. PiOS VLC will do this for you, or if you want demo progs look at my drmu (DRM render) or hello_wayland (Wayland render) projects.

juanitotc, 2025-05-11:
Thanks.

I built drmu against rpi-ffmpeg test/7.1.1/main, but I get this:

```
./hello_drmu /mnt/mmcblk0p2/media/video/*hevc*
...
Decoder hevc does not support device type drm.
No h/w format found - trying s/w
```

jc-kynesim, 2025-05-12:
OK. I've got to assume you aren't using PiOS so some of the below may be wrong / inapplicable.

There are two likely failures (1) you haven't managed to build a working ffmpeg or (2) it isn't installed somewhere that drmu finds when building / running. It can be very easy to have a mix of libraries and get the wrong one.

Easiest test for a working ffmpeg - boot to a command line so you don't have a GUI trying to grab the display then:

ffmpeg -v trace -no_cvt_hw -hwaccel drm -i <your stream> -show_all 1 -f vout_drm -

That should at the very least generate some debug. If you get moving pictures then hurrah! we have a working ffmpeg and we can move on to wondering why drmu isn't finding it.
Assuming not then send me the debug (or at least the first few hundred lines till it becomes repetitive trying to play the stream if it gets that far).
For "correct" ffmpeg config I suggest you look at pi-utils/conf_native.sh (script I use for test setup on PiOS) - assuming you aren't on PiOS then install paths may want changing.

juanitotc, 2025-05-12:
Sorry, I should have said that I'm using piCore64 on an RPi5.

I've built a working rpi-ffmpeg test/7.1.1/main in that it will play files without using hardware acceleration - there's no other version of ffmpeg loaded so it's using the correct libav* and friends.

In the ./hello_drmu example above the video plays using sw acceleration.

log file attached

[log.tar.gz](https://github.com/user-attachments/files/20159516/log.tar.gz)

jc-kynesim, 2025-05-12:
Thanks for the log. FFmpeg isn't even trying to use hwaccel, so it probably isn't compiled in successfully. After running configure in the list of stuff configured you should find "hevc_v4l2request" under "Enabled hwaccels", if you don't then you haven't compiled in the support you want. If you do have it then I'm confused.

I'd recommend adding these options to your config and see if it helps:
```
 --enable-sand
 --enable-v4l2-request
 --enable-libdrm
```

juanitotc, 2025-05-13:
Thanks - I didn't know about --enable-sand and --enable-v4l2-request

Some progress, but the video freezes after a few seconds.

I thought the RPi allocated memory themselves now without the need for specifying cma memory, for example dtoverlay=vc4-kms-v3d,cma-128, or is this something else?

```
./hello_drmu /mnt/mmcblk0p2/media/video/*hevc*
../drmu/drmu_xlease.c:55:get_lease_fd: index 116 screen 0 rem 1
../drmu/drmu_xlease.c:81:get_lease_fd: output[0/2] 66: conn 0/0 crtc 64
../drmu/drmu_xlease.c:127:get_lease_fd: get_lease_fd OK: fd=4
Pixfmt after init: yuv420p / (null)
[hevc @ 0x7fff3c0218d0] Hwaccel V4L2 HEVC stateless V4; devices: /dev/media1,/dev/video19; buffers: src DMABuf, dst DMABuf; swfmt=rpi4_8
Failed to alloc 3112960 from dma-heap(fd=15): 12 (Cannot allocate memory)
[hevc @ 0x7fff3c2552b0] v4l2_request_hevc_start_frame: Failed to get dst buffer
[hevc @ 0x7fff3c2552b0] Error parsing NAL unit #0.
Failed to alloc 3112960 from dma-heap(fd=15): 12 (Cannot allocate memory)
[hevc @ 0x7fff3c0218d0] v4l2_request_hevc_start_frame: Failed to get dst buffer
[hevc @ 0x7fff3c0218d0] Error parsing NAL unit #0.
Error during decoding
Failed to alloc 3112960 from dma-heap(fd=15): 12 (Cannot allocate memory)
[hevc @ 0x7fff3c1971a0] v4l2_request_hevc_start_frame: Failed to get dst buffer
[hevc @ 0x7fff3c1971a0] Error parsing NAL unit #0.
Error while decoding
../drmu/drmu_poll.c:244:evt_polltask_cb: evt_polltask_cb: Timeout
```

juanitotc, 2025-05-13:
Here's the log from:
```
ffmpeg -v trace -no_cvt_hw -hwaccel drm -i /mnt/mmcblk0p2/media/video/*hevc* -show_all 1 -f vout_drm - 2>&1 | tee -a log
```

[log.tar.gz](https://github.com/user-attachments/files/20186268/log.tar.gz)

jc-kynesim, 2025-05-13:
OK - now we are into possible OS differences. FFmpeg will allocate dmabufs from whatever /dev/dma_heap/vidbuf_cached is pointing to (it has fallbacks if that doesn't exist but it appeared to find that). On a Pi5 running PiOS that points to /dev/dma_heap/system which will give it non-contiguous dmabuf buffers - at least on PiOS there are as many of these as you have memory; on a Pi4, because it has no IOMMU on video decode, it would point to  /dev/dma_heap/linux,cma which allocates out of CMA which is, as you note, a limited resource and ffmpeg is quite greedy. What does your vidbuf_cached point to?
It may also be worth checking dmesg for clues.

juanitotc, 2025-05-13:
```
ls -l /dev/dma_heap/vidbuf_cached
lrwxrwxrwx    1 root     root             9 May 13 14:15 /dev/dma_heap/vidbuf_cached -> linux,cma
```

..so setting "dtoverlay=vc4-kms-v3d,cma-256" fixes things and the video plays hevc hardware accelerated :)

If I build rpi-ffmpeg test/7.1.1/main statically linked, will hello_drmu link to it statically?

Thanks for the help

popcornmix, 2025-05-13:
If you change it to:
```
ls -l  /dev/dma_heap/vidbuf_cached
lrwxrwxrwx 1 root root 6 May 12 16:38 /dev/dma_heap/vidbuf_cached -> system
```
You won't need to increase cma (and that is the default behaviour of RPiOS on Pi5).

jc-kynesim, 2025-05-13:
hello_drmu uses meson to build. meson uses magic. Give it a go, if it works, then it works if not then build shared ffmpeg libs. I've never tested without having shared libs.

juanitotc, 2025-05-13:
It works - now hello_drmu can live alongside the standard ffmpeg :)

closed 2025-05-13 by juanitotc