jc-kynesim/rpi-ffmpeg #99: hardware decode h265 on Rpi5

harryqt, 2025-05-23:
Hi, I want to use hardware decode h265 on Rpi5, but the best I can use is `-hwaccel drm`, eg:

```bash
ffmpeg -y -hide_banner -hwaccel drm -i "rtsp://user:pass@192.168.5.63:554/Streaming/Channels/101" -t 1 -f null -
```

But I don't think its hardware decoding.. what am I missing or doing wrong?

Similar question: https://forums.raspberrypi.com/viewtopic.php?t=387861

jc-kynesim, 2025-05-23:
Difficult to say with the info provided. Add "-v trace" to the start of the line, try again and send me the log. Are you using PiOS?
It is worth noting that ffmpeg has an implicit h/w -> s/w conversion by default so it can actually get at and use the frame. Given you are just dropping the frames on the floor this isn't needed, add "-no_cvt_hw" at the start of the command line to prevent it - this will reduce processing dramatically if h/w decode is actually working.

harryqt, 2025-05-24:
Problem solved..!

After upgrading to [Frigate - 0.16.0 Beta 2](https://github.com/blakeblackshear/frigate/releases) (based on Debian Bookworm), I received `FFmpeg` version`5.1.6-0+deb12u1+rpt3`.

```sh
harry@rpi5:~ $ ffmpeg -y -hide_banner -hwaccel drm -rtsp_transport tcp -i "rtsp://uri" -t 1 -f null -
Input #0, rtsp, from 'rtsp://uri':
  Metadata:
    title           : Media Presentation
  Duration: N/A, start: 0.040000, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 1520x2688, 25 fps, 25 tbr, 90k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> wrapped_avframe (native))
Press [q] to stop, [?] for help
[hevc @ 0x555576df31b0] Hwaccel V4L2 HEVC stateless V4; devices: /dev/media1,/dev/video19; buffers: src DMABuf, dst DMABuf; swfmt=rpi4_8
Output #0, null, to 'pipe:':
  Metadata:
    title           : Media Presentation
    encoder         : Lavf59.27.100
  Stream #0:0: Video: wrapped_avframe, yuv420p(pc, bt709, progressive), 1520x2688, q=2-31, 200 kb/s, 25 fps, 25 tbn
    Metadata:
      encoder         : Lavc59.37.100 wrapped_avframe
frame=   26 fps=0.0 q=-0.0 Lsize=N/A time=00:00:01.00 bitrate=N/A speed=2.35x
video:12kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
harry@rpi5:~ $
```
Now FFmpeg's CPU consumption reduced to just 3-4% from 15-17% 🤞

closed 2025-05-24 by harryqt