jc-kynesim/rpi-ffmpeg #100: rtsp h265hw dec -> x264 hw enc on rpi4 leads to "non-existing PPS 0 referenced" on rtsp ffmpeg rx side

Hoernchen, 2025-06-02:
I'm trying to hw transcode a cropped stream coming from a IP cam on my rpi4, current rpi os (Linux raspberrypi 6.12.27-v8+ #1876 SMP PREEMPT Tue May  6 15:12:28 BST 2025 aarch64 GNU/Linux), your "dev/7.1.1/update_bit_size_1" branch because I was hoping a newer ffmpeg would fix the issue, but the same thing as with stock ffmpeg happens:
```
 ✘  ~  ffmpeg -y -probesize 300000 -hwaccel drm -i "rtsp://admin:123456@192.168.2.50/stream1" -c:v h264_v4l2m2m  -vf "crop=1024:1024:1419:616" -f rtsp rtsp://192.168.2.222:8554/reenckue
ffmpeg version 3e3136e Copyright (c) 2000-2025 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14+deb12u1)
  configuration: --enable-sand --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-libdrm
  libavutil      59. 39.100 / 59. 39.100
  libavcodec     61. 19.101 / 61. 19.101
  libavformat    61.  7.100 / 61.  7.100
  libavdevice    61.  3.100 / 61.  3.100
  libavfilter    10.  4.100 / 10.  4.100
  libswscale      8.  3.100 /  8.  3.100
  libswresample   5.  3.100 /  5.  3.100
Input #0, rtsp, from 'rtsp://admin:123456@192.168.2.50/stream1':
  Metadata:
    title           : HTMS
    comment         : stream1
  Duration: N/A, start: 0.067000, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 3840x2160, 15 fps, 15 tbr, 90k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> h264 (h264_v4l2m2m))
Press [q] to stop, [?] for help
[hevc @ 0x557421dee0] Hwaccel V4L2 HEVC stateless V4; devices: /dev/media0,/dev/video19; buffers: src DMABuf, dst DMABuf; swfmt=rpi4_8
[h264_v4l2m2m @ 0x5574049db0]  <<< v4l2_encode_init: fmt=0/-1
[h264_v4l2m2m @ 0x5574049db0] Using device /dev/video11
[h264_v4l2m2m @ 0x5574049db0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x5574049db0] requesting formats: output=YU12/yuv420p capture=H264/none
Output #0, rtsp, to 'rtsp://192.168.2.222:8554/reenckue':
  Metadata:
    title           : HTMS
    comment         : stream1
    encoder         : Lavf61.7.100
  Stream #0:0: Video: h264, yuv420p(tv, bt709, progressive), 1024x1024, q=2-31, 200 kb/s, 15 fps, 90k tbn
      Metadata:
        encoder         : Lavc61.19.101 h264_v4l2m2m
frame= 3284 fps= 15 q=-0.0 size=N/A time=00:03:38.86 bitrate=N/A speed=   1x
```

This breaks frigates ffmpeg trying to receive the transocoded cropped video:
```
[.... spam continues ...]
2025-06-02 21:46:33.089072664  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [h264 @ 0x632caeabe0c0] decode_slice_header error
2025-06-02 21:46:33.089251147  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [h264 @ 0x632caeabe0c0] no frame!
2025-06-02 21:46:33.089257639  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [h264 @ 0x632caeabe0c0] non-existing PPS 0 referenced
2025-06-02 21:46:33.089442235  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   :     Last message repeated 1 times
2025-06-02 21:46:33.089555598  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [h264 @ 0x632caeabe0c0] decode_slice_header error
2025-06-02 21:46:33.089694780  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [h264 @ 0x632caeabe0c0] no frame!
2025-06-02 21:46:33.089776556  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [rtsp @ 0x632caeab03c0] Could not find codec parameters for stream 0 (Video: h264, none): unspecified size
2025-06-02 21:46:33.089916292  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : Consider increasing the value for the 'analyzeduration' (0) and 'probesize' (5000000) options
2025-06-02 21:46:33.089998136  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [segment @ 0x632caebbb8c0] dimensions not set
2025-06-02 21:46:33.090123716  [2025-06-02 21:46:33] ffmpeg.kuchr.detect            ERROR   : [out#0/segment @ 0x632caeae3440] Could not write header (incorrect codec parameters ?): Invalid argument
```

However, doing the same thing on a rpi5 (Linux pi5cam 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux, minus the hw x264 enc) does work fine:
```
➜  ~ ffmpeg -hwaccel drm -rtsp_transport tcp -i "rtsp://admin:123456@192.168.2.50/stream1" -c:v libx264 -preset fast  -vf "crop=1024:1024:1419:616"  -f rtsp rtsp://192.168.2.222:8554/reenckue
ffmpeg version 5.1.6-0+deb12u1+rpt3 Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr --extra-version=0+deb12u1+rpt3 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sand --enable-sdl2 --disable-sndio --enable-libjxl --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
Input #0, rtsp, from 'rtsp://admin:123456@192.168.2.50/stream1':
  Metadata:
    title           : HTMS
    comment         : stream1
  Duration: N/A, start: 0.143000, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 3840x2160, 15 fps, 15 tbr, 90k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (hevc (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[hevc @ 0x55a13ee2e0] Hwaccel V4L2 HEVC stateless V4; devices: /dev/media3,/dev/video19; buffers: src DMABuf, dst DMABuf; swfmt=rpi4_8
[libx264 @ 0x55a1404b10] using cpu capabilities: ARMv8 NEON
[libx264 @ 0x55a1404b10] profile High, level 3.2, 4:2:0, 8-bit
[libx264 @ 0x55a1404b10] 264 - core 164 r3095 baee400 - H.264/MPEG-4 AVC codec - Copyleft 2003-2022 - http://www.videolan.org/x264.html - options: cabac=1 ref=2 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=6 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=6 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=1 keyint=250 keyint_min=7 scenecut=40 intra_refresh=0 rc_lookahead=30 rc=crf mbtree=1 crf=23.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, rtsp, to 'rtsp://192.168.2.222:8554/reenckue':
  Metadata:
    title           : HTMS
    comment         : stream1
    encoder         : Lavf59.27.100
  Stream #0:0: Video: h264, yuv420p(pc, bt709, progressive), 1024x1024, q=2-31, 7 fps, 90k tbn
    Metadata:
      encoder         : Lavc59.37.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/0 buffer size: 0 vbv_delay: N/A
frame=   42 fps= 12 q=24.0 size=N/A time=-00:00:00.14 bitrate=N/A speed=N/A
```

So it looks like the hw x264 enc in the pi4 is at fault, or the stream is being mishandled by ffmpeg?

jc-kynesim, 2025-06-03:
OK - if you want a more up to date branch of mine I suggest test/7.1.1/main as a relatively stable one. the one you've picked, whilst I think it is good, is my development branch for testing our attempts to upstream the H265 kernel driver and so may sometimes be deliberately "broken".

Having said that - it should work for what you want. When you say "frigates ffmpeg" is that another Pi running our ffmpeg or is it something else? (It shouldn't matter much as I haven't touched RTSP rx  but I'd like to know for testing.)  Do you know the version of the ffmpeg it is running? The last time someone complained about this it was because when running with h/w decode ffmpeg chose (entirely legally / correctly) to encode the H.264 SPS/PPS in the RTSP headers rather than in the stream and the receiver didn't like that.

spookyfirehorse, 2025-06-06:
hi and hello !!!

os bookworm with ffmpeg 5.1.6   and ./configure 

 --prefix=/usr --extra-version=0+deb12u1+rpt3 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sand --enable-sdl2 --disable-sndio --enable-libjxl --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a53 --arch=armv7 --disable-armv5te --disable-armv6t2 --disable-cuda --disable-vaapi --disable-vdpau --disable-htmlpages --disable-manpages --disable-podpages --disable-txtpages --enable-nonfree --enable-libfdk-aac --enable-vout-drm --enable-vout-egl --disable-static --enable-vulkan --disable-static

i stream from my rpi3 with this script

rpicam-vid  -b 1000000 --autofocus-mode continuous  --denoise cdn_off \
  --codec libav --libav-format mpegts  --brightness 0.1 --contrast 1.0 \
  --sharpness   1.0  --level 4.2 --framerate 25 --width 1536 --height 864 \
  --audio-device=plughw:CARD=Device,DEV=0   --audio-bitrate=96kbps \
  --audio-codec libopus  --audio-channels 1 --libav-audio 1 --audio-source alsa \
  -t 0  -n --inline -o  - | ffmpeg -fflags \
  +genpts+nobuffer+igndts+discardcorrupt -flags low_delay \
  -vcodec h264_v4l2m2m -i - -metadata title='MOON' -codec copy \
  -f rtsp -rtsp_transport tcp  rtsp://"user:password"@"localhost:8558"/mystream

and on another rpi3 armhf receving

os bookworm with ffmpeg 5.1.6 ./configure

 --prefix=/usr --extra-version=0+deb12u1+rpt3 --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sand --enable-sdl2 --disable-sndio --enable-libjxl --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a53 --arch=armv7 --disable-armv5te --disable-armv6t2 --disable-cuda --disable-vaapi --disable-vdpau --disable-htmlpages --disable-manpages --disable-podpages --disable-txtpages --enable-nonfree --enable-libfdk-aac --enable-vout-drm --enable-vout-egl --disable-static --enable-vulkan --disable-static

and this script

ffmpeg  -an -c:v h264_v4l2m2m  -fflags +nobuffer   -rtsp_transport tcp   -i    rtsp://"user:password"@10.0.0.12:8558/mystream -c:v h264_v4l2m2m -filter:v "crop=640:360:320:180"   -rtsp_transport tcp   -f rtsp   rtsp://"user:password"@localhost:8556/mystream

Working

##################################################################

os trixie ffmpeg 7.1.1 rpi 4  64 bit aarch not working

ffmpeg  -an -c:v h264_v4l2m2m  -fflags +nobuffer   -rtsp_transport tcp   -i    rtsp://"user:password"@10.0.0.12:8558/mystream -c:v h264_v4l2m2m -filter:v "crop=640:360:320:180"   -rtsp_transport tcp   -f rtsp   rtsp://"user:password"@localhost:8556/mystream

ffmpeg version n7.1-256-g6dbf87aefd-0+deb12u1+rpt3 Copyright (c) 2000-2025 the FFmpeg developers
  built with gcc 14 (Debian 14.2.0-19)
  configuration: --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig  [... cut ...]
  Stream #0:0 -> #0:0 (h264 (h264_v4l2m2m) -> h264 (h264_v4l2m2m))
Press [q] to stop, [?] for help
[h264_v4l2m2m @ 0x55a0340de0]  <<< v4l2_encode_init: fmt=0/-1
[h264_v4l2m2m @ 0x55a0340de0] Using device /dev/video11
[h264_v4l2m2m @ 0x55a0340de0] driver 'bcm2835-codec' on card 'bcm2835-codec-encode' in mplane mode
[h264_v4l2m2m @ 0x55a0340de0] requesting formats: output=YU12/yuv420p capture=H264/none

no error or something

and no streaming just nothing

i dont try it with ffmpeg 7.1.1 on bookworm

jc-kynesim, 2025-06-09:
As I haven't tried anything on Trixie yet if you'd like to try either 5.1.6 on Trixie or 7.1.1 on bookworm that would be helpful.
Ta

spookyfirehorse, 2025-06-12:
no problems running like charm

jc-kynesim, 2025-06-12:
Closing as there doesn't appear to be an ffmpeg bug here. Feel free to reopen if you have new evidence.

closed 2025-06-12 by jc-kynesim

Hoernchen, 2025-06-20:
Sorry for the delay, I finally got around to giving this another try, and yes, apparently go2rtc was at fault (in the "receiver" role), it works as expected with mediamtx, or by providing a custom exec line for go2rtc. The command line examples provided by @spookyfirehorse work fine, too. That being said, I think it would help if the rpi docs would contain basic go2rtc/mediamtx examples with ffmpeg -> ffplay or something like that, a "known good" example, since that is what apparently everyone tries to do. but actual up to date information is scattered all over the forum and gh issues...

lurch, 2025-06-20:
> I think it would help if the rpi docs would contain basic go2rtc/mediamtx examples with ffmpeg -> ffplay or something like that

I dunno if i's useful or not, as I know very little about video decoding or streaming, but there _are_ some instructions related to MediaMTX at https://www.raspberrypi.com/documentation/computers/camera_software.html#webrtc

Hoernchen, 2025-06-21:
> I dunno if i's useful or not, as I know very little about video decoding or streaming, but there _are_ some instructions related to MediaMTX at https://www.raspberrypi.com/documentation/computers/camera_software.html#webrtc

Yes, but those don't really cover a full example with hw encoding/decoding, and some of that changed, so if you google it you end up wiht OMX which was all the rage until 2020 (?) and that information rather useless, currently it's all v4l stateless.. And then there were some changes to the camera api. So the issue is that a lot of the information you can find does not really apply any more to a current setup. There is no concise example for gstreamer with hw blocks either, and only gstreamer can use the mjpeg hardware blocks (at least according to a forum post), and so on..